magicbug/Cloudlog #2456: Update problem

do4bmw, 2023-08-31:
when i try to update via the update script i get this output

cloudlog:/var/www/html# ./update_cloudlog.sh 
Cloudlog update started
-----------------------
Using /var/www/html as working directory
Fetching changes from remote...
Fetched latest changes successfully
Fast-forwarding to HEAD (bleeding-edge)...
Updating 98bd2ea2..26d726a2
error: Your local changes to the following files would be overwritten by merge:
        update_cloudlog.sh
Please commit your changes or stash them before you merge.
Aborting
Fast-forward failed; stopping here.
[image]

phl0, 2023-08-31:
> > error: Your local changes to the following files would be overwritten by merge:
> > update_cloudlog.sh

> what i am doing wrong ?

As it says: There are local changes in update_cloudlog.sh. You probably changes that file.

do4bmw, 2023-08-31:
This is the file , i dont know what changed 

```
#!/bin/bash
# Cloudlog update script
#
# Pulls changes from Git remote and re-sets appropriate directory ownership.
# Can be run manually or on a schedule using Cron.
#
# Please check the DIR_OWNERSHIP variable carefully to make sure it is
# appropriately set for your system!

# The user and group that own the CLOUDLOG_SUBDIR directories. Passed to 'chown' as-is.
DIR_OWNERSHIP="root:www-data"
# The list of directories that need to have ownership restored after a git pull
declare -a CLOUDLOG_SUBDIRS=("application/config" "assets" "backup" "updates" "uploads")
# The name of the Git remote to fetch/pull from
GIT_REMOTE="origin"
# If true, pull from the HEAD of the configured origin, otherwise the latest tag
BLEEDING_EDGE="true"
# If true, restore directory ownership on CLOUDLOG_SUBDIRS after a git pull
RESTORE_OWNERSHIP="true"

check_working_dir() {
        # Quick sanity check to make sure that pwd looks like a Cloudlog install
        if [[ ! -d "$(pwd)/application" ]]
        then
                echo "$(pwd) doesn't look like a Cloudlog install directory! Stopping here."
                exit 1
        fi
}

fast_forward_to_tag() {
        # Find the latest tag on the current branch
        # See https://git-scm.com/docs/git-describe for details on retrieving tags
        LATEST_TAG=$(git describe --tags --abbrev=0)
        echo "Fast-forwarding to latest tag: $LATEST_TAG..."
        if git pull $GIT_REMOTE $LATEST_TAG ; then
                echo "Fast-forward finished successfully."
        else
                echo "Fast-forward failed; stopping here."
                exit 2
        fi
}

fast_forward_to_head() {
        echo "Fast-forwarding to HEAD (bleeding-edge)..."
        if git pull $GIT_REMOTE ; then
                echo "Fast-forward finished successfully".
        else
                echo "Fast-forward failed; stopping here."
                exit 3
        fi
}

restore_ownership() {
        for dir in "${CLOUDLOG_SUBDIRS[@]}"; do
                echo "Setting ownership as $DIR_OWNERSHIP on $dir"
                chown -R $DIR_OWNERSHIP $(pwd)/$dir
        done
}

echo "Cloudlog update started"
echo "-----------------------"
echo "Using $(pwd) as working directory"

check_working_dir

# Fetch the latest changes from the master branch
echo "Fetching changes from remote..."
if git fetch $GIT_REMOTE --tags ; then
        echo "Fetched latest changes successfully"
else
        echo "Fetch failed; stopping here."
        exit 4
fi

if [ "$BLEEDING_EDGE" = true ]; then
        fast_forward_to_head
else
        fast_forward_to_tag
fi

if [ "$RESTORE_OWNERSHIP" = true ]; then
        restore_ownership
fi

echo "Cloudlog update finished"

```

phl0, 2023-08-31:
Do a 
```
git diff
```
That will show the differences.

do4bmw, 2023-08-31:
[image]

and the list goes on with more callsigns

do4bmw, 2023-08-31:
ah and with git diff update_cloudlog.sh 

i get this output

[image]

phl0, 2023-08-31:
Set the permission back with
```
chmod 0644 update_cloudlog.sh
```
And see if the update works. Start it with `bash update_cloudlog.sh`

do4bmw, 2023-08-31:
not realy

[image]

do4bmw, 2023-08-31:
back to 755

```
root@cloudlog:/var/www/html# ./update_cloudlog.sh
Cloudlog update started
-----------------------
Using /var/www/html as working directory
Fetching changes from remote...
Fetched latest changes successfully
Fast-forwarding to HEAD (bleeding-edge)...
Updating 98bd2ea2..26d726a2
error: Your local changes to the following files would be overwritten by merge:
        update_cloudlog.sh
Please commit your changes or stash them before you merge.
Aborting
Fast-forward failed; stopping here.
root@cloudlog:/var/www/html#

```

phl0, 2023-08-31:
> not realy
> 
> [image]

I asked you to execute `bash update_cloudlog.sh`.

do4bmw, 2023-08-31:
> > not realy
> > [image]
> 
> I asked you to execute `bash update_cloudlog.sh`.

oh , okay ... thats it ... it now worked 

many thanks

closed 2023-08-31 by do4bmw