magicbug/Cloudlog #2459: Select QSL requests by category (Electronic, Bureau, Direct)

m0urs, 2023-09-01:
I have again a problem with QSL management and Cloudlog.

We had the discussion regarding setting QSL sent status to "Requested" vice "Queued". I used that in the past to differentiate between Bureau requests (set to "Requested") and electronic requests by mail (set to "Queued").

The outcome of the discussion was that we will only use "Requested" from now on, because you do have the possibility to differentiate the request with the "Sent Method". 

So I used now only "Requested" and now wanted to create the QSL cards.

Here are my problems:

- If I am using the "Print Requested QSLs" option I do get all my requests listed and I can see if these are "Electronic", "Direct" or "Bureau" requests. That is fine.

However, if I then select e.g. only the electronic cards and click on "Export requested QSLs to ADIF file" I do get ALL requested QSLs exported (as the name of the button is saying). I do not have the possibility to create an ADIF file with only the selected ones. So I cannot use this way.

- Then I tried the "Advanced Logbook"

Here I can select all QSOs with status "QSL sent" set to "Requested", but I do not see the "Sent Method" so I am unable to select e.g. only the direct or only the Bureau QSLs.

At least I did not find that option (maybe I missed something?)

There is a possibility to select "QSL via" and that gives me a selection of "B", "D","E", and "M" but if I do set it to "E" e.g. I do not get all electronic requests. "QSL via" should also means something different ...

So my question is now:

What do I miss here? How can I now get an ADIF file with just the QSLs I want to have, means selected by "Direct", "Bureau" and "Electronic"?

HB9HIL, 2023-10-03:
@magicbug This looks like a bug-like problem to me. 

M0URS is right. The QSL via Dropdown shouldn't give the possibility to select the Sent Method. Rather it should be a Clear Text field to search for the "Via" QSL Managers. 

In addition to that there should be the option to filter for "QSL Sent" by Bureau, Direct or Electronic, aswell for "QSL Received" by Bureau, Direct or Electronic

What do you think?

magicbug, 2023-10-04:
Yes VIA needs to be a text box

HB9HIL, 2023-10-04:
> It is a combo-box, which uses a list, so you can have a dropdown and write free text. That is not the issue here. Rather it is that B/E/D/M should search col_qsl_sent_via, and the free text should search col_qsl_via. So it should really be split up.
> 
> The current solution searches only col_qsl_via now.

Working on it :-)

HB9HIL, 2023-10-05:
Dear @m0urs 

We added your requested filter functionality. 👍 

Well be published with the next release 😄 🥳

m0urs, 2023-10-05:
Great! Thank you very much. Will be happy to test later!

HB9HIL, 2023-10-05:
> Great! Thank you very much. Will be happy to test later!

Is in the `dev` branch. Please feedback 😄

m0urs, 2023-10-05:
@HB9HIL 

Thank you very much. That is exactly what I was looking for. Great work!

PR #2541 (2023-10-05): Logbookadvanced enhanced QSL Filter
Hello there

I try to fix #2459 and try to add follwing 

- adding two new Filter Options "QSL Sent Method" and "QSL Recv Method" to make search even more precise
- removed datalist from "QSL via" Form because "QSL via" should be a search for QSL Managers and not the Method 

My actual Problem is, that I can't get the filtering to get working. 

I added in `application/models/Logbookadvanced_model.php` 

```
if ($searchCriteria['qslSentMethod'] !== '') {
	$condition = "COL_QSL_SENT_VIA like ?";
	$conditions[] = $condition;
	$binding[] = $searchCriteria['qslS [... cut]
closed 2023-10-05 by m0urs