magicbug/Cloudlog #2468: Database Error Occurred Error Number: 1366

M5JFS, 2023-09-06:
Trying to log a QSO but getting this error
A Database Error Occurred
Error Number: 1366

Incorrect double value: '' for column `m5jfscloudlog`.`TABLE_HRD_CONTACTS_V01`.`COL_DISTANCE` at row 1

INSERT INTO `TABLE_HRD_CONTACTS_V01` (`COL_TIME_ON`, `COL_TIME_OFF`, `COL_CALL`, `COL_BAND`, `COL_BAND_RX`, `COL_FREQ`, `COL_MODE`, `COL_SUBMODE`, `COL_RST_RCVD`, `COL_RST_SENT`, `COL_NAME`, `COL_COMMENT`, `COL_SAT_NAME`, `COL_SAT_MODE`, `COL_COUNTRY`, `COL_CONT`, `COL_QSLSDATE`, `COL_QSLRDATE`, `COL_QSL_SENT`, `COL_QSL_RCVD`, `COL_QSL_SENT_VIA`, `COL_QSL_RCVD_VIA`, `COL_QSL_VIA`, `COL_QSLMSG`, `COL_OPERATOR`, `COL_QTH`, `COL_PROP_MODE`, `COL_IOTA`, `COL_DISTANCE`, `COL_FREQ_RX`, `COL_ANT_AZ`, `COL_ANT_EL`, `COL_A_INDEX`, `COL_AGE`, `COL_TEN_TEN`, `COL_TX_PWR`, `COL_STX`, `COL_SRX`, `COL_STX_STRING`, `COL_SRX_STRING`, `COL_CONTEST_ID`, `COL_NR_BURSTS`, `COL_NR_PINGS`, `COL_MAX_BURSTS`, `COL_K_INDEX`, `COL_SFI`, `COL_RX_PWR`, `COL_LAT`, `COL_LON`, `COL_DXCC`, `COL_CQZ`, `COL_STATE`, `COL_CNTY`, `COL_SOTA_REF`, `COL_WWFF_REF`, `COL_POTA_REF`, `COL_SIG`, `COL_SIG_INFO`, `COL_DARC_DOK`, `COL_NOTES`, `station_id`, `COL_MY_GRIDSQUARE`, `COL_HRDLOG_QSO_UPLOAD_STATUS`, `COL_QRZCOM_QSO_UPLOAD_STATUS`, `COL_MY_CITY`, `COL_MY_IOTA`, `COL_MY_SOTA_REF`, `COL_MY_WWFF_REF`, `COL_MY_POTA_REF`, `COL_STATION_CALLSIGN`, `COL_MY_DXCC`, `COL_MY_COUNTRY`, `COL_MY_CNTY`, `COL_MY_CQ_ZONE`, `COL_MY_ITU_ZONE`, `COL_GRIDSQUARE`, `COL_EQSL_QSL_SENT`, `COL_EQSL_QSL_RCVD`, `COL_LOTW_QSL_SENT`, `COL_LOTW_QSL_RCVD`) VALUES ('2023-08-26 21:49', '2023-08-26 21:49', 'DJ0MY', '2m', '70cm', 145965000, 'SSB', NULL, '59', '59', '', '', 'RS-44', 'V/U', 'Federal Republic Of Germany', 'EU', NULL, NULL, 'N', 'N', '', NULL, '', '', 'M5JFS', '', 'SAT', '', '', 435640000, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, '', '', NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, '230', '14', '', NULL, '', '', '', '', '', '', '', '1', 'IO90KU', 'N', 'N', 'PORTCHESTER', 'EU-005', '', '', '', 'M5JFS', '223', 'ENGLAND', '', '14', '27', 'JN49', 'N', 'N', 'N', 'N')

Filename: models/Logbook_model.php

Line Number: 540

Looks like it is trying to put an empty string '' into the COL_DISTANCE column

When I enter the QSO, the grid square is not populated automatically so I keyed it in manually (JN49) and the border of the Gridsquare box turns green. The error occurs after hitting Save QSO

[image]

phl0, 2023-09-07:
Can you check if the javascript console spits any error messages?

magicbug, 2023-09-07:
Also what version of Cloudlog and branch

M5JFS, 2023-09-07:
Annoyingly, despite being a solid fail (I tried logging this QSO a few times and each time I got the same error) but the problem is no longer occurring. 

for what it's worth, version is 2.4.8 
Branch | dev
Commit | 8400654d
Tag | 2.4.7-113-g8400654d
Last Fetch | Wednesday, 06-Sep-23 23:00:03 GMT+0000

closed 2023-09-20 by magicbug