magicbug/Cloudlog #2469: Advanced Search showing duplicate rows

magicbug, 2023-09-07:
Can't figure this one out, I'm seeing duplicate rows of the same QSO which even has the same primary id

[image]

I had the same on the standard logbook view and adding a distinct seemed to resolve it but I cant seem todo the same for this.

Thoughts @phl0 @int2001 @AndreasK79 

Happening on both dev and master

int2001, 2023-09-07:
do you have two certs with the same call, @magicbug ?

magicbug, 2023-09-07:
Nope

magicbug, 2023-09-07:
Resolved, seems the lotw_users table had duplicate entries!

closed 2023-09-07 by magicbug

int2001, 2023-09-07:
we are going to create a mig-script, which puts a unique-idx on lotw_users.call.
i think root-cause was a broken cronjob or sth. like that.

AndreasK79, 2023-09-07:
Happy you figured it out. Table should be emptied when cron is run, so a bit strange. Unique index will fix it though.