magicbug/Cloudlog #2473: Feature request: quick way to log a new QSO

jholubinka, 2023-09-11:
Hello,

Adding a new entry in the log should be an essential feature of a logging app. In Cloudlog it's hidden in the menu - it needs 2 clicks (or 3 on mobile view) and it's not even in the first place. I made a quick and dirty hack which replaces the search box with a form that collects a callsign and redirects to the "Post QSO" page with populated callsign. I'm fully aware that this patch is awful, but I hope it will give you a little bit of inspiration for doing it as it should be ;)

Kind regards,
Jakub SP9VIK

```diff
diff --git a/application/views/interface_assets/footer.php b/application/views/interface_assets/footer.php
index 262abec5..00569f46 100644
--- a/application/views/interface_assets/footer.php
+++ b/application/views/interface_assets/footer.php
@@ -1052,6 +1052,7 @@ $(document).on('keypress',function(e) {
     });

      // Callsign always has focus on load
+      $("#callsign").focusout();  // Force reload of a list of previous QSOs
       $("#callsign").focus();

       if ( ! manual ) {
diff --git a/application/views/interface_assets/header.php b/application/views/interface_assets/header.php
index 234a479f..0b3e6736 100644
--- a/application/views/interface_assets/header.php
+++ b/application/views/interface_assets/header.php
@@ -195,10 +195,11 @@
     </ul>

      <?php if($this->optionslib->get_option('global_search') != "false" || $this->session->userdata('user_type') >= 2) { ?>
-               <form method="post" action="<?php echo site_url('search'); ?>" class="form-inline">
-               <input class="form-control mr-sm-2" id="nav-bar-search-input" type="search" name="callsign" placeholder="<?php echo lang('menu_search_text'); ?>" aria-label="Search">
+               <form method="get" action="<?php echo site_url('qso'); ?>" class="form-inline">
+               <input class="form-control mr-sm-2" id="nav-bar-search-input" type="search" name="callsign" placeholder="New QSO" aria-label="New QSO">
+                <input type="hidden" name="manual" value="1">

-               <button title="<?php echo lang('menu_search_button'); ?>" class="btn btn-outline-success my-2 my-sm-0" type="submit"><i class="fas fa-search"></i></button>
+               <button title="Add QSO" class="btn btn-outline-success my-2 my-sm-0" type="submit"><i class="fas fa-plus"></i></button>
                </form>
        <?php } ?>

diff --git a/application/views/qso/index.php b/application/views/qso/index.php
index 5c257555..cc962bab 100755
--- a/application/views/qso/index.php
+++ b/application/views/qso/index.php
@@ -74,7 +74,7 @@
               <div class="form-row">
                 <div class="form-group col-md-9">
                   <label for="callsign"><?php echo lang('gen_hamradio_callsign'); ?></label><?php if ($this->optionslib->get_option('dxcache_url') != '') { ?>&nbsp;<i id="check_cluster" data-toggle="tooltip" data-original-title="Search DXCluster for latest Spot" class="fas fa-search"></i> <?php } ?>
-                  <input type="text" class="form-control" id="callsign" name="callsign" required>
+                  <input type="text" class="form-control" id="callsign" name="callsign" value="<?php echo $_GET['callsign']; ?>" required>
                   <small id="callsign_info" class="badge badge-secondary"></small> <a id="lotw_link"><small id="lotw_info" class="badge badge-success"></small></a>
                 </div>
                 <div class="form-group col-md-3 align-self-center">

```

petrkr, 2023-09-11:
it goes to Live QSO with ticking timer or static Post QSO where you still need add time by hand?

jholubinka, 2023-09-12:
> it goes to Live QSO with ticking timer or static Post QSO where you still need add time by hand?

It goes to manual mode, but if you want it live, just change 1 to 0 in the line ``<input type="hidden" name="manual" value="1">``.

HB9HIL, 2023-10-08:
Maybe something like this?

[image]

Let me know what you think. Creating a PR

HB9HIL, 2023-10-08:
> it goes to Live QSO with ticking timer or static Post QSO where you still need add time by hand?

My PR goes to **Live Mode**. Because I think If you want to log a Post-QSO you may have time enough to navigate with the menu. Quicklog should be for logging NOW a Live QSO. 

What do you think?

petrkr, 2023-10-08:
How it would looks like on mobile? Mostly those qso are somewhere in terain without laptop

HB9HIL, 2023-10-09:
[image]

Mobile is no Problem
Maybe the Position between QSO Table and "QSO Breakdown" would be better at the top. Maybe I can write something responsive

jholubinka, 2023-10-09:
> Mobile is no Problem Maybe the Position between QSO Table and "QSO Breakdown" would be better at the top. Maybe I can write something responsive

In my opinion this should be as accessible as it can be. Ideally, it should be on every page to allow user to start logging no matter whether they're browsing their log, or doing something else.

HB9HIL, 2023-10-09:
> In my opinion this should be as accessible as it can be. Ideally, it should be on every page to allow user to start logging no matter whether they're browsing their log, or doing something else.

@magicbug What do you think about replacing the search button with this Quicklog Field as the initial idea of @jholubinka ? Maybe no bad idea because Search is nothing time critical but logging is sometimes.

jholubinka, 2023-10-09:
On one hand, we could make something more versatile, such as something like that:

[image]

On the other hand, I've got a strange feeling that using one input box and two associated buttons that do different things is completely against many UI design guidelines. As far as I know, CloudLog is the only logbook software that still follows them, and it would be really nice not to break it.

_Sidenote: I exchanged few emails with the current maintainer of Swisslog, and we both agreed that people can be either good programmers or good HAM operators, but never both. ;)_

magicbug, 2023-10-09:
> On one hand, we could make something more versatile, such as something like that:
> 
> [image]

While it might solve one issue it will create many others so you are quite correct this won't be accepted as a PR

> Sidenote: I exchanged few emails with the current maintainer of Swisslog, and we both agreed that people can be either
> good programmers or good HAM operators, but never both. ;)

I am not sure I agree personally, the biggest thing you have to watch for is bloat which ham radio apps seem to love

HB9HIL, 2023-10-16:
@jholubinka 
I've been thinking about your feature request, and I think it's going to be pretty tough to make sure it works well on every page.

One option would be to put the Quicklog field in the header, but that's kind of like your two-in-one idea, which goes against the usual design rules.

Another idea could be to add a setting in the user options that lets you choose whether you want to see the Quicklog field instead of the search field. However, this would mess with the design a bit and make it a bit harder to find the search function (we'd need to add another option to the dropdown).

Right now, I'm not sure what the best way is to meet your request without messing up the basic design principles.

HB9HIL, 2023-11-05:
Dear @jholubinka 

we discussed that feature and found that this will be implemented as you mentioned but with a User Option to enable or disable it. 

The feature was merged today in `dev` branch with PR #2554 and is ready to test. This Feature will be in `master` branch with the next stable release. 

Please test it in the `dev` branch and give some feedback. Thanks for your suggestion and using Cloudlog 😄

jholubinka, 2023-11-06:
> Please test it in the `dev` branch and give some feedback. Thanks for your suggestion and using Cloudlog 😄

Works as expected :) The only issue I've found is that when default action is to log a QSO, the text field is still treated as a search box. It's a minor detail, very easy to miss, but it's visible on the phone. A fix for that is obvious.

[image]

HB9HIL, 2023-11-06:
> Works as expected :) The only issue I've found is that when default action is to log a QSO, the text field is still treated as a search box. It's a minor detail, very easy to miss, but it's visible [...]

Which browser do you use on Android?

HB9HIL, 2023-11-06:
Got it. Will be fixed

HB9HIL, 2023-11-06:
There is another bug when pressing the reset button. 

Trying to fix that.

HB9HIL, 2023-11-14:
@jholubinka The new version 2.5.1 was just released. Everything should now working fine. Please try and give feedback. This helps. 

Changed the way how Callsign Data goes from the Quicklog Field to the QSO window. If you have some problems first try Ctrl + F5 (on Win) or Shift+Command+R (on Mac) to reload the page with clearing the cache.

HB9HIL, 2023-11-14:
And please close this issue if you think it fits now 👍

jholubinka, 2023-11-17:
> Please try and give feedback. This helps.

The only feedback I can give you is a positive one :) 
Thank you and 73!

closed 2023-11-17 by jholubinka