magicbug/Cloudlog #2485: Routing Error due to volume mounts in docker container

filliphy, 2023-09-14:
**Describe the bug**
After installation page of 2m0sql/cloudlog docker container (when details as Gridsquare locator, and database name are input), unable to continue as `Unable to determine what should be displayed. A default route has not been specified in the routing file.` error is shown.

**To Reproduce**
As per the wiki:
1. Create persistent volumes (config, backup and uploads) - I used bind folders on my docker host.
2. run docker command `docker run -v cloudlog-config:/var/www/html/application/config -v cloudlog-backup:/var/www/html/application/backup -v cloudlog-uploads:/var/www/html/application/uploads --name cloudlog -p 8086:80 2m0sql/cloudlog`
3. Start container and complete information - **Confguration Settings** & **Database settings**
4. Press Install Button
5. Error displayed as above `Unable to determine what should be displayed....`

**Expected behaviour**
Login screen for Cloudlog should show up at  login with default details.

**To fix behaviour**
1. After install phase, files _config.php_ and _database.php_ are created.
2. Stop docker container
3. Bind **config.php** and **database.php** in the config directory _only_ instead of the complete config directory. and restart container.
4. will work as expected.

rayslinky, 2023-09-14:
I've also run into this. I've been running cloudlog for a while but recently switched over from the waarlandit/cloudlog image to 2m0sql/cloudlog. I'm using a bind mount to my local ssd instead of a docker volume. The files in my config dir were seemingly from my initial install, and I found that bind mounting just the 2 that a user usually changes sorted my issues too.

magicbug, 2023-09-20:
I'm not a big docker person so if this is something I can automate do let me know

rayslinky, 2023-09-26:
If there are only 2 files that we expect users to change, perhaps those could be put into a sub-directory that can be mapped in, ie config/user or config/instance.

magicbug, 2023-09-28:
@iu2frl can we implement these?

iu2frl, 2023-09-28:
Did any of these files, permissions or behavior changed recently?

magicbug, 2023-12-12:
Lets be clear as a project I have decided not to provide docker support going forward, as this isn't code related if you need help with setting up server components use discussions

closed 2023-12-12 by magicbug