magicbug/Cloudlog #2496: Upload to LoTW does not work

DL4AN, 2023-09-18:
The new QSOs do not get uploaded to the LoTW and keep being "red" in Logbook-Overview.
Neither GUI nor Cronjob with curl .../index.php/lotw/lotw_upload work.

The status of the LoTW certificate is shown as "Valid. Not synced"

Cloudlog has the newest version from github.

magicbug, 2023-09-18:
Is this a brand new install? what version of PHP

DL4AN, 2023-09-18:
Not a new install, it is in use since ~2 years. I've tested it with PHP versions 7.4 and 8.2 with the no effect.

DL4AN, 2023-09-18:
Update - the reason is an unset cert_dxcc_id and therefore "continue" in this statement in application/controllers/Lotw.php
`
                               // If Station Profile has no LoTW Cert continue on.
                               if(!isset($data['lotw_cert_info']->cert_dxcc_id)) {
                                        continue;
                                }
`

DL4AN, 2023-09-18:
Works now - the DXCC country in the station profile was "Germany DL/D (Deleted)"
After changing it to "Federal Republic of Germany - DL" the LoTW sync works again.

closed 2023-09-18 by DL4AN