magicbug/Cloudlog #2516: Error when clicking "Logout"

bhoglan, 2023-09-25:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When logged into the cloudlog dash, when I click on my account name in the upper right then click Logout at the bottom I'm presented with an error page consisting of 3 PHP errors. PHPmyadmin container has been stopped. I don't see any log output for the main or mysql containers aside from just the page access records.

**To Reproduce**
Steps to reproduce the behaviour:
1. When logged into cloudlog click your account name in the upper right
2. Click on Logout
3. The resulting page is the error page

**Expected behaviour**
The console should log out and the user should be presented with either a "Logged out" screen or the login screen again.

**Screenshots**
[image]

**Server:**
- OS: OS: Ubuntu Server 22.04.3
- Docker version: 24.0.6 build ed223bc
- Docker Compose version: 2.21.0
- Cloudlog image versions:
mysql             latest    b2013ac99101   3 days ago    577MB
phpmyadmin        latest    61dcc026c415   4 days ago    562MB
2m0sql/cloudlog   latest    d7ade8554e52   12 days ago   794MB

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.5.2 (ARM64)
 - Browser: Firefox
 - Version: 117.0.1
 - 2nd Browser: Chrome
 - 2nd Browser Version: 117.0.5938.92 

**Additional context**
Access logs from main container:
`cloudlog-main  | 192.168.10.217 - - [25/Sep/2023:19:11:26 +0000] "GET /index.php/visitor/radio_display_component HTTP/1.1" 200 338 "http://cloudlog.mydomain.com:8086/index.php/dashboard" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:109.0) Gecko/20100101 Firefox/117.0"
cloudlog-main  | 192.168.10.217 - - [25/Sep/2023:19:11:27 +0000] "GET /index.php/user/logout HTTP/1.1" 200 761 "http://cloudlog.mydomain.com:8086/index.php/dashboard" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:109.0) Gecko/20100101 Firefox/117.0"
cloudlog-main  | 192.168.10.217 - - [25/Sep/2023:19:11:29 +0000] "GET /index.php/dashboard HTTP/1.1" 307 427 "http://cloudlog.mydomain.com:8086/index.php/user/login" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:109.0) Gecko/20100101 Firefox/117.0"
cloudlog-main  | 192.168.10.217 - - [25/Sep/2023:19:11:30 +0000] "GET /index.php/user/login HTTP/1.1" 200 6898 "http://cloudlog.mydomain.com:8086/index.php/user/login" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:109.0) Gecko/20100101 Firefox/117.0"`

bhoglan, 2023-09-25:
I spoke too soon. Here's another set of PHP errors. This one is encountered when trying to change the theme under Admin>Global Options>Appearance>Theme.
I select a theme and when I click Save I'm presented with these PHP errors.
[image]

nathanharig, 2023-09-28:
Just want to add I am having the same issues, I am using the dockerized container via a portainer install as found in the wiki, and in addition to random PHP errors I also remain unable to find/disable developer mode.

magicbug, 2023-09-28:
as per the wiki you can turn off developer mode by changing the environment variable in index.php to production

magicbug, 2023-09-28:
@iu2frl any ideaas

nathanharig, 2023-09-28:
Is this the wiki page that is being referenced?
https://github.com/magicbug/Cloudlog/wiki/Debug? The wiki navigation, to
someone coming completely new to this, is not pointing me to the right
result of how to edit index.php in a dockerized container image.

Nathan A. Harig

phl0, 2023-09-28:
> I spoke too soon. Here's another set of PHP errors. This one is encountered when trying to change the theme under Admin>Global Options>Appearance>Theme. I select a theme and when I click Save I'm presented with these PHP errors.

This is simply a bug. Fix proposed in https://github.com/magicbug/Cloudlog/pull/2520.

phl0, 2023-09-28:
> **Describe the bug** When logged into the cloudlog dash, when I click on my account name in the upper right then click Logout at the bottom I'm presented with an error page consisting of 3 PHP errors. 

Seems you are running a newer version of PHP here which requires a different parameter handling. To solve this I back-ported the session function for cookie destroy from recent CI 3 code. Would like to receive some feedback if this works in https://github.com/magicbug/Cloudlog/pull/2521 before un-drafting and merging the code.

iu2frl, 2023-09-28:
The docker image was recently upgraded to a 8.X version as I was told it was fully compatible

phl0, 2023-09-28:
> The docker image was recently upgraded to a 8.X version as I was told it was fully compatible

We can never be 100% sure. :)

iu2frl, 2023-09-28:
Of course, otherwise there would be no more need for software developers 😆

phl0, 2023-09-28:
> The docker image was recently upgraded to a 8.X version as I was told it was fully compatible

Which version exactly? Could not reproduce the error with 8.1.23 here...

iu2frl, 2023-09-28:
Version in dockerfile: FROM php:8.2-apache

bhoglan, 2023-09-28:
> Is this the wiki page that is being referenced? https://github.com/magicbug/Cloudlog/wiki/Debug? The wiki navigation, to someone coming completely new to this, is not pointing me to the right result of how to edit index.php in a dockerized container image. Nathan A. Harig
> […](#)

@nathanharig I don't know if this is the right solution but here's how I do it. These docker images are minimized to reduce any extra overhead so they remove editors as it's not really expected to maintain the images/containers that way.
But you can do this:
1. Enter the main container: `docker exec -it cloudlog-main /bin/bash` 
2. Once in the container (The prompt will look something like this: `root@894cffa033cd:/var/www/html#` ) you can update the package cache (really you're populating the cache for the first time) `apt update`.
3. Then you can install your preferred command line editor. In my case it's Vim using `apt install vim` or `apt install nano` or whatever your heart desires.
4. Edit the file using your editor. The line in question (in my container but I don't think yours would be any different) /var/www/html/index.php line 57.
5. Restart the container (for funsies, I don't think the server actually needs to be restarted, it should just read the new file on page load and be fine)

magicbug, 2023-10-04:
We have added some code changes into the latest release that might resolve this so do let us know

bhoglan, 2023-10-04:
@magicbug Yes the issue appears resolved. I can logout and I can change the theme without PHP errors. Thanks for the help!

closed 2023-10-04 by magicbug