magicbug/Cloudlog #2517: Feature: HTML meta to make mobile act as app

petrkr, 2023-09-25:
There is some meta tag, which will allow make webpage act as application, so if in chrome is used "Create Shortcut" function, it will hide address bar and show it in fullscreen. For example in android.

Now if I will do shortcut, web browser opens with URL address bar.

Insipartion could be taken in nextcloud for example.

I will try ask and add such meta tag, but keeping this issue/task if someone know it "out of box" it could be faster

HB9HIL, 2023-10-03:
Duplicate? #2294

petrkr, 2023-10-04:
Maybe? If that feature is called "progressive" app. But he mention also offline mode in that progressive app. I only suggest make it act as app, not as webpage.

But someone told me, there is some better cache if it is like. So especially on slower connections it could loads faster

magicbug, 2023-10-04:
The browser is already caching based off server settings so I don't see how adding a few meta tags will improve that, most of the interaction in Cloudlogs with a database in a remote location

magicbug, 2023-10-04:
> There is some meta tag, which will allow make webpage act as application, so if in chrome is used "Create Shortcut" function, it will hide address bar and show it in fullscreen. For example in android.

Which meta tags would you like us to consider adding?

HB9HIL, 2023-10-04:
Probably he means 
```
apple-mobile-web-app-capable
```
So the website acts like an App but just on gui side. This Meta Tag works for Safari and Chrome Mobile Browsers

magicbug, 2023-10-04:
OK think safe to say this is part of Progressive Web App so closing as its a duplicate

closed 2023-10-04 by magicbug

HB9HIL, 2023-11-14:
@petrkr 

The new Version 2.5.1 was just released! It now supports acting like a Progressive Web App but for now just online-only.

For iPhones:

Tap on the Share Button in Safari and add Cloudlog to your Homescreen. It should act like a App and will be shown in Fullscreen.

On Android Phones you can choose "Install Cloudlog" from the Browser Menu.

Same on Desktop Browsers (only Chrome and Edge, Safari and Firefox do not support PWA). On the right side of the URL Line should now appear the Option "install".

Try that and give some feedback.

dg9vh, 2023-11-14:
works like a charm (using it on dev-branch for the last days) here on chrome os-device.

HB9HIL, 2023-11-14:
> works like a charm (using it on dev-branch for the last days) here on chrome os-device.

good to hear 👍