magicbug/Cloudlog #2519: Advanced Logbook - An Error Occurred

Michael-AUG, 2023-09-27:
Hi

A while back now I was importing an ADI file into my CloudLog instance and something went wrong. I don't know exactly what; but now, whenever I click on 'Advanced Log', I get this error message display:

[image]

Is there something I can do at the server end to reset this/fix it? I would go and fiddle around but I don't want to lose all my logs!

Thanks
Michael GM5AUG

phl0, 2023-09-27:
Can you have a look at the network tab and let us know what the network requests for table content says?

Michael-AUG, 2023-09-27:
[image]
Is this what you need to see? Or something else?

phl0, 2023-09-27:
The details of the packet resulting in the error 500 at the bottom is of interest. Can you show the details and the response here?

Michael-AUG, 2023-09-27:
[image]
Thanks Florian.

phl0, 2023-09-27:
Yeah and here the response tab please?

Michael-AUG, 2023-09-27:
Response tab empty... screenshots:
[image]
[image]
[image]

phl0, 2023-09-27:
Could you please enable development mode in index.php and re-do that? Looking for the error message which is hidden in prod mode ...

Michael-AUG, 2023-09-27:
[image]
[image]
[image]
Is this right?

phl0, 2023-09-27:
Set 

```
define('ENVIRONMENT', 'development');
```

in index.php. Again open dev tools of your browser, call the advanced search and show the response tab of the packet producing the error 500 pls.

Michael-AUG, 2023-09-27:
Sorry to be thick Florian... do you mean type `define...`  in the Developer Tools - Console? Or edit the index.php file?

magicbug, 2023-09-27:
You'll need to edit index.php set the environment to development

phl0, 2023-09-27:
No worries. Edit index.php. Find the line containing `define('ENVIRONMENT')` and change production to development here.
Then enable developer tools in your browser. Select the Network tab and go to advanced search again. Then show the response tab of the packet causing the error 500 please.

Michael-AUG, 2023-09-27:
Yup - gotcha. Was just about to delete that message. Thanks Peter. I'm on it, Florian.

Michael-AUG, 2023-09-27:
[image]
OK... ^

phl0, 2023-09-27:
Yeah. This helps (I guess at least). Let me check.

phl0, 2023-09-27:
Can you supply the adif file which produced the error? Would be apprechiated via mail. My mail is ok on QRZ.com :)

Michael-AUG, 2023-09-27:
No I'm afraid I can't. I stupidly deleted it when it wasn't working :{

magicbug, 2023-09-27:
Export of your logbook from Cloudlog might be the next best thing

phl0, 2023-09-27:
Hmmm I set BAND to null for a test QSO. But does not produce an error in logbook advanced ....

phl0, 2023-09-27:
Maybe you can provide an SQL dump of your DB/table?

phl0, 2023-09-27:
@Michael-AUG Which version of Cloudlog are you using btw?

phl0, 2023-10-03:
Ping

Michael-AUG, 2023-10-04:
Hi guys. Sorry for radio silence on this one.

> Export of your logbook from Cloudlog might be the next best thing

Sorry - being thick. Do you mean an ADIF? Or an export from the server?

> Which version of CloudLog are you using btw?

I'm using v 2.3.3 

> Maybe you can provide an SQL dump of your DB/table?

Could you talk me through how to do this please?

Thanks everyone - really appreciate your help.
Michael

magicbug, 2023-10-04:
> I'm using v 2.3.3

First I'd suggest updating to see if it resolves itself, but if not yes ADIF export via Cloudlog would be a good start.

Michael-AUG, 2023-10-04:
OK - I'll get to work on that now.

ryandzink, 2023-10-09:
I am seeing the same issue, but I don't remember ever having a failed ADIF import. What info can I help provide that might help get to a root cause?

phl0, 2023-10-09:
> I am seeing the same issue, but I don't remember ever having a failed ADIF import. What info can I help provide that might help get to a root cause?

Which version are you using? And does it show the same error concerning band in the details?

ryandzink, 2023-10-10:
@phl0 I am using v2.4.10

Here is the response on the HTTP 500 that I am seeing:

```
An uncaught Exception was encountered

Type: ParseError

Message: syntax error, unexpected 'string' (T_STRING), expecting function (T_FUNCTION) or const (T_CONST)

Filename: /opt/bitnami/apache2/htdocs/src/QSLManager/QSO.php

Line Number: 12
```

phl0, 2023-10-10:
> @phl0 I am using v2.4.10

Thats good so far :) Is that happening on specific filters or on first call directly? Is it maybe one (or more) specific QSOs that cause the error? If so you do an SQL dump showing the stored data for that/those QSOs?

phl0, 2023-10-10:
> @phl0 I am using v2.4.10
> 
> Here is the response on the HTTP 500 that I am seeing:

Oh and which PHP/SQL versions are you using?

ryandzink, 2023-10-12:
I'm using MySQL 5.7 and PHP 7.3.15. It happens even on first load of the page, before executing any searches.

I'm happy to query the TABLE_HRD_CONTACTS_V01 table but it's got several thousand QSOs so I might need something specific to look for.

AndreasK79, 2023-10-12:
You need at least PHP 7.4. The search is executed on page load, so that is why you see it. Try upgrading PHP and see if that fixes the problem.

closed 2023-11-15 by AndreasK79