magicbug/Cloudlog #2529: Winkey stopped working

ok1cdj, 2023-10-01:
There is not possible to select port when you clicked on Connect button, also there is problem to save macros. Dialog with Macros Saved appears then click Save and nothing. Can you check please...

AndreasK79, 2023-10-04:
Yeah, seems some problems there. #2540 is a work in progress to fix.

AndreasK79, 2023-10-04:
@ok1cdj could you test latest dev to see if that fixes the problem?

magicbug, 2023-10-04:
Should help but I think it will still need some work to reconnect to the comport on page reload

GregTroar, 2023-11-27:
The issue I have is that the macros are not being saved at all.
You put a name and the text:

CQ

CQ de F4BPO F4BPO K

Click Save then close
Re open the window everything is blank

It is not saving macros at all

closed 2025-05-28 by magicbug