magicbug/Cloudlog #2539: New logbook + location, QSOs not visible in overview

kconkas, 2023-10-03:
**Describe the bug**
I have created a new logbook, a new location and added several QSOs into the new log. Now I can only see the newly added QSOs in "Logbook" -> "Advanced" if I filter by "De" (my new call sign), whereas "Logbook" -> "Overview" shows no QSOs and presents a message "No logbooks were found. You need to define a logbook under Station Logbooks!".

In the log I observed the following error while attempting to view the logbook overview:
```
ERROR - 2023-10-03 22:06:00 --> Severity: error --> Exception: Call to a member function result() on array /var/www/html/cloudlog/application/controllers/Logbook.php 566
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Station Logbooks
2. Create a new logbook
3. Set new logbook as active
4. Go to Station Locations
5. Create a new location
6. Set new location as active
7. Add some QSOs

**Expected behaviour**
To be able to see the newly added QSOs in the new logbook.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.6.5
 - Browser: Firefox 117.0.1

**Additional context**
This bug was observed while running cloudlog 2.4.9. I have since upgraded to cloudlog 2.4.10 to no avail.

I also observed that for other logbooks/locations when I switch to them, I can see the details of the chosen logbook and location, for example:
[image]

Whereas for newly added logbook + location I can't:
[image]

I don't know if it is relevant but from this new logbook/location pair I even managed to successfully upload to LoTW (with the right certificate).

magicbug, 2023-10-03:
Did you add the location to the station logbook?

kconkas, 2023-10-04:
@magicbug yes I did (or at least I think I did). I created a new logbook, made it active and then created a location. I was expecting this to link them as it did for several other logbooks/locations I've got active. Is there a way to check this?

magicbug, 2023-10-04:
Edit the Station Logbook and see "Linked"

petrkr, 2023-10-04:
> Edit the Station Logbook and see "Linked"

I had this problem too, maybe some feature improvement to select "first" or "default" station to link during adding ?

closed 2023-10-04 by kconkas