magicbug/Cloudlog #2542: Feature: Contest support moon contest

petrkr, 2023-10-04:
Contest does not have dynamic fields. only few presets.

During moon contest (https://ok2vbz.waypoint.cz/mc/) there are fields:

CALL, RST, Serial (počínaje 001), grid (6-letter) and QTH name.

for now I put QTH to Comment and retype it for log. Will be handy to have this field there aswell. I think it is "Exchange" field ?

But there is only Serial + exchange or Serial + gridsquare,, there is not combination serial + grid + exchange (or qth)

HB9HIL, 2023-10-05:
@magicbug Maybe it would be a cool option to make "Exchange Type" to a Checkbox List so the hams can choose which Exchanges they need. Similar to the Options Box for the Advanced Logbook so choose which columns to be shown in the table.

AndreasK79, 2023-10-05:
@HB9HIL it kinda does that now, since visible columns change according to what you choose in the dropdown. But feel free to expand on it.

HB9HIL, 2023-10-05:
I'll check that out.

petrkr, 2023-10-05:
> @HB9HIL it kinda does that now, since visible columns change according to what you choose in the dropdown. But feel free to expand on it.

Or have some common presets, and last one "custom" which can show mentioned check boxes.

HB9HIL, 2023-10-05:
> > @HB9HIL it kinda does that now, since visible columns change according to what you choose in the dropdown. But feel free to expand on it.
> 
> Or have some common presets, and last one "custom" which can show mentioned check boxes.

Good idea :)