magicbug/Cloudlog #2543: Feature: Support EDI contest format

petrkr, 2023-10-04:
Moon contest need EDI format, because it support more fields than cabrillo format (for example QTH name).

Would be nice to have direct export to this format aswell

magicbug, 2023-10-17:
Do you have any information about this format?

HB9HIL, 2023-10-17:
> Moon contest need EDI format, because it support more fields than cabrillo format (for example QTH name).
> 
> Would be nice to have direct export to this format aswell

@petrkr This one? [](https://www.ok2kkw.com/ediformat.htm)

petrkr, 2023-10-17:
Yes, i can send sample, which I exported later

abarrau, 2023-11-05:
for information, this feature is available on contest plugin

closed 2024-02-01 by magicbug

petrkr, 2025-05-04:
> for information, this feature is available on contest plugin
> 
> [image]

after year I again started VKV contests, but I still not see EDI export and this is closed. You said it is in contest plugin, but my contest page looks different than on screenshot