magicbug/Cloudlog #2549: Unable to setup cloudlog without error

fattredd, 2023-10-06:
Seems as though there and php errors during the initial setup.

**Describe the bug**
Following the [docker stack](https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack) instructions, upon clicking the "Install" button a php error occurs in `helpers/url_helper.php`

The error is as follows:
```
A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/application/migrations/139_modify_eQSL_url.php:1)

Filename: helpers/url_helper.php

Line Number: 564

**To Reproduce**
Steps to reproduce the behaviour:
1.  [docker stack](https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack) instructions
2. Run `docker compose up -d`
3. Fill out the init form
4. Click Install

**Expected behaviour**
The setup completes

fattredd, 2023-10-06:
I've also been seeing several other `A PHP Error was encountered`. I'm unsure of how related they are:  
  
  
When uploading my `.p12` certificate:
```

A PHP Error was encountered

Severity: Warning

Message: Undefined array key "cert"

Filename: controllers/Lotw.php

Line Number: 407
A PHP Error was encountered

Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/system/core/Exceptions.php:283)

Filename: helpers/url_helper.php

Line Number: 564
```

HB9HIL, 2023-10-08:
> Seems as though there and php errors during the initial setup.
> 
> **Describe the bug** Following the [docker stack](https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack) instructions, upon clicking the "Install" button a php error occurs in `helpers/url_helper.php`
> 
> The error is as follows:
> 
> ```
> A PHP Error was encountered
> Severity: Warning
> 
> Message: Cannot modify header information - headers already sent by (output started at /var/www/html/application/migrations/139_modify_eQSL_url.php:1)
> 
> Filename: helpers/url_helper.php
> 
> Line Number: 564
> ```
> 
> Interestingly, opening the page in a new tab allows me to log in as though everything were fine.
> 
> **To Reproduce** Steps to reproduce the behaviour:
> 
>     1. [docker stack](https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack) instructions
> 
>     2. Run `docker compose up -d`
> 
>     3. Fill out the init form
> 
>     4. Click Install
> 
> 
> **Expected behaviour** The setup completes

I could not replicate this php errors with the basic setup described in the wiki. 
On which system do you run the stack?
Did you made any other changes to the 'docker-compose.yml' then just the Passwords?

fattredd, 2023-10-17:
Sorry for the delay. I'm running on aarch64 (raspberry pi 4, 64-bit).  
  
For the sake of ensuring I didn't muck something up, I tried again with a freshly copied `docker-compose.yml` and deleted the existing volumes. For testing I didn't bother changing anything in the yaml -- not even the passwords.  
  
I start the stack with `docker compose up` so I could watch the output. After waiting a bit for mysql to initialize, I navigate to `http://localhost:8086/`.  

I fill out the database settings sections with `hostname: cloudlog-mysql`, `username: cloudlog`, `password: cloudlog-pass`, and `database name: cloudlog`. I click the Install button and get a page that says:  
  
```
A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/application/migrations/139_modify_eQSL_url.php:1)

Filename: core/Common.php

Line Number: 575

An uncaught Exception was encountered
Type: mysqli_sql_exception

Message: Table 'cloudlog.users' doesn't exist

Filename: /var/www/html/system/database/drivers/mysqli/mysqli_driver.php

Line Number: 301
```

And here's the `docker compose` output. I've thrown it in a gist so it doesn't clutter the page:  
https://gist.github.com/fattredd/7bd83d2aa7eeff658e33cad83cd507f0

HB9HIL, 2023-10-17:
> Sorry for the delay. I'm running on aarch64 (raspberry pi 4, 64-bit).
>
> For the sake of ensuring I didn't muck something up, I tried again with a freshly copied `docker-compose.yml` and deleted the [...]

Which Kernel do you use on the rapsi?

You can find out with `uname -r`

HB9HIL, 2023-10-17:
@fattredd I was able to reproduce your error. Seems like there is a Problem with ArmV8. I would recommend to install cloudlog on the raspi the traditional way like [the wiki says](https://github.com/magicbug/Cloudlog/wiki/Installation). 

@iu2frl Could you check that?

iu2frl, 2023-10-17:
I don't think is anything on the raspberry itself but rather a missing table on the database, did you run through the "install" procedure?

HB9HIL, 2023-10-17:
> I don't think is anything on the raspberry itself but rather a missing table on the database, did you run through the "install" procedure?

I did the exact same steps of a basic stack install on a 

**raspberry pi 4 - os light 32 bit --> error**
[image]

**raspberry pi 4 - os light 64 bit --> error**
same as @fattredd already mentioned above
```
A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/application/migrations/139_modify_eQSL_url.php:1)

Filename: core/Common.php

Line Number: 575

An uncaught Exception was encountered
Type: mysqli_sql_exception

Message: Table 'cloudlog.users' doesn't exist

Filename: /var/www/html/system/database/drivers/mysqli/mysqli_driver.php

Line Number: 301
```

basic debian 12 vm on amd64 --> no errors, installation successfull with problems

iu2frl, 2023-10-17:
This is your issue: Message: Table 'cloudlog.users' doesn't exist, anything else can be ignored. How are you connecting to the database? Did you create users and permissions to access the database? Can you reach the database itself using any third party client?

HB9HIL, 2023-10-17:
> This is your issue: Message: Table 'cloudlog.users' doesn't exist, anything else can be ignored. How are you connecting to the database? Did you create users and permissions to access the database? 

I did nothing different as in the debian amd64 machine. And there it worked without any problems. 

> Can you reach the database itself using any third party client?

not from outside of docker because I didn't change the `docker-compose.yml` 
If I could reach it from inside docker I didn't test.

f5mzn, 2023-10-27:
Hi - Same error message here, runing the install procedure as described on the wiki. This is on OSX, with Docker Desktop. The `cloudlog.users` table is well created, though. 73

iu2frl, 2023-10-27:
I tried on develop on a completely blank machine and i got a different error:

```
A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/application/migrations/139_modify_eQSL_url.php:1)

Filename: helpers/url_helper.php

Line Number: 564
```

HB9HIL, 2023-10-27:
> I tried on develop on a completely blank machine and i got a different error:
>
> ```
> A PHP Error was encountered
> Severity: Warning
>
> Message: Cannot modify header information - headers already sent by [...]

I created a debian 12 machine on amd64 and do not have any issues. Which kind of machine did u use?

magicbug, 2023-12-12:
Lets be clear as a project I have decided not to provide docker support going forward, as this isn't code related if you need help with setting up server components use discussions

closed 2023-12-12 by magicbug