magicbug/Cloudlog #2555: Database Error with Typo in Time Field

ntilley905, 2023-10-09:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Entering a QSO with an extra digit in the time produces a database error.

**To Reproduce**
Steps to reproduce the behaviour:
1. In the time section on a post QSO, enter a typo in the field such as "718:07"
2. Finish the remainder of the log and submit it
3. Error described below occurs

**Expected behaviour**
Input sanitation to verify that the entry can be accepted by the software.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux (Pop OS)
 - Browser: Firefox
 - Version: 118.0.1 (64-bit)

**Additional context**
Error log:
```

A Database Error Occurred

Error Number: 1292

Incorrect datetime value: '2023-10-08 718:07' for column `cloudlog`.`TABLE_HRD_CONTACTS_V01`.`COL_TIME_ON` at row 1

INSERT INTO `TABLE_HRD_CONTACTS_V01` (`COL_TIME_ON`, `COL_TIME_OFF`, `COL_CALL`, `COL_BAND`, `COL_BAND_RX`, `COL_FREQ`, `COL_MODE`, `COL_SUBMODE`, `COL_RST_RCVD`, `COL_RST_SENT`, `COL_NAME`, `COL_COMMENT`, `COL_SAT_NAME`, `COL_SAT_MODE`, `COL_COUNTRY`, `COL_CONT`, `COL_QSLSDATE`, `COL_QSLRDATE`, `COL_QSL_SENT`, `COL_QSL_RCVD`, `COL_QSL_SENT_VIA`, `COL_QSL_RCVD_VIA`, `COL_QSL_VIA`, `COL_QSLMSG`, `COL_OPERATOR`, `COL_QTH`, `COL_PROP_MODE`, `COL_IOTA`, `COL_DISTANCE`, `COL_FREQ_RX`, `COL_ANT_AZ`, `COL_ANT_EL`, `COL_A_INDEX`, `COL_AGE`, `COL_TEN_TEN`, `COL_TX_PWR`, `COL_STX`, `COL_SRX`, `COL_STX_STRING`, `COL_SRX_STRING`, `COL_CONTEST_ID`, `COL_NR_BURSTS`, `COL_NR_PINGS`, `COL_MAX_BURSTS`, `COL_K_INDEX`, `COL_SFI`, `COL_RX_PWR`, `COL_LAT`, `COL_LON`, `COL_DXCC`, `COL_CQZ`, `COL_STATE`, `COL_CNTY`, `COL_SOTA_REF`, `COL_WWFF_REF`, `COL_POTA_REF`, `COL_SIG`, `COL_SIG_INFO`, `COL_DARC_DOK`, `COL_NOTES`, `station_id`, `COL_MY_GRIDSQUARE`, `COL_HRDLOG_QSO_UPLOAD_STATUS`, `COL_QRZCOM_QSO_UPLOAD_STATUS`, `COL_MY_CITY`, `COL_MY_IOTA`, `COL_MY_SOTA_REF`, `COL_MY_WWFF_REF`, `COL_MY_POTA_REF`, `COL_STATION_CALLSIGN`, `COL_MY_DXCC`, `COL_MY_COUNTRY`, `COL_MY_CNTY`, `COL_MY_CQ_ZONE`, `COL_MY_ITU_ZONE`, `COL_GRIDSQUARE`, `COL_LOTW_QSL_SENT`, `COL_LOTW_QSL_RCVD`) VALUES ('2023-10-08 718:07', '2023-10-08 718:07', 'KN6KJA', '20m', '', 14033000, 'CW', NULL, '579', '579', 'Arthur R', '', '', '', 'United States Of America', 'NA', NULL, NULL, 'N', 'N', '', NULL, '', '', 'N7NWT', 'Mount Etna', '', '', '1037.9', 0, NULL, NULL, NULL, NULL, NULL, '5', NULL, NULL, '', '', NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, '291', '5', 'CA', 'CA,Alameda', 'W6/NS-399', '', '', '', '', '', '', '49', 'CN96EW', 'N', 'N', 'PALISADES PEAK', '', 'W7W/RS-008', '', 'K-0052', 'N7NWT', '291', 'UNITED STATES OF AMERICA', 'PIERCE', '3', '6', 'CM99NR', 'N', 'N')

Filename: models/Logbook_model.php

Line Number: 548
```

HB9HIL, 2023-10-09:
confirm

phl0, 2023-10-25:
Fix proposes in https://github.com/magicbug/Cloudlog/pull/2620.

closed 2023-10-25 by magicbug