magicbug/Cloudlog #2568: [PHP 8.1] foreach() argument must be of type array|object, null given /mnt/www/application/models/Dxcluster_model.php 41

pa8s, 2023-10-15:
**Describe the bug**
While using the DXCluster Bandmap , the log file reports following error:
```
ERROR - 2023-10-15 19:08:10 --> Severity: Warning --> foreach() argument must be of type array|object, null given /mnt/www/applicati
on/models/Dxcluster_model.php 41
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Select the Bandmap from the menu.
2. Select a radio in the dropdown menu.
3. Check the most recent logfile.

**Expected behaviour**
No error in logfile.

**Server**
OS: FreeBSD 13.1
Server: nginx/1.24.0
PHP: 8.1.24
Git branch: dev
Commit: https://github.com/magicbug/Cloudlog/commit/ab75d5807df87a46b5c34d5dd601c7f3fa90af81
Tag: [2.4.10-120-gab75d580](https://github.com/magicbug/Cloudlog/releases/tag/2.4.10)

int2001, 2023-10-16:
Can't reproduce. Which band? Or does this always happen?

int2001, 2023-10-16:
interesting. what options did you set at Admin/GlobalOptions/DXCluster?
and: do you run your CL at home or hosted?

can you try to call the following URL via curl/browser **from the machine where Cloudlog runs**?
https://dxc.jo30.de/dxcache/spots/10m

please post the result here.

closed 2023-10-16 by pa8s