magicbug/Cloudlog #2577: Tab-complete look-ups lock up web interface

tgreaves, 2023-10-17:
**Describe the bug**
Logging a Live QSO, then hitting TAB to do a look-up on the ID, results in the web interface freezing.  apache2 process goes to 100% CPU.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Log a Live QSO
2. Enter a call ID
3. Hit TAB
4. See error

**Desktop (please complete the following information):**
 - OS: MacOS Sonoma 14.0
 - Browser: Microsoft Edge
 - Version 118.0.2088.46 (Official build) (arm64)

Error log extract:

[Tue Oct 17 12:07:28.851996 2023] [php7:error] [pid 450] [client 192.168.1.61:53477] PHP Fatal
 error:  Maximum execution time of 30 seconds exceeded in /var/www/html/application/controller
s/Lookup.php on line 84, referer: http://192.168.1.14/index.php/qso?manual=0

phl0, 2023-10-17:
Can you please outline which environment Cloudlog runs in? Which OS, which web server, which database.

magicbug, 2023-10-17:
If its self hosted some information on hardware too would be useful, sounds like a poorly configured setup

tgreaves, 2023-10-17:
Looks similar in behaviour to this issue:

https://github.com/magicbug/Cloudlog/discussions/2574

The VM is currently running Ubuntu 20.04.5 and dedicated to CloudLog.  It has 2 VCPUs and 512 megs of RAM allocated, of which 350 Megs is still marked as free.  The underlying host CPU is 8 x Intel(R) Core(TM) i5-1035G4 CPU @ 1.10GHz and sits about 5-6% use most of the time. Local MySQL database. All other CloudLog functionality is working fine.

tgreaves, 2023-10-17:
[image]

tgreaves, 2023-10-17:
Thanks for the reply.

I've done a bit more research, and this is working just fine if I use Firefox instead.  I TAB out of the Callsign box, and it does the lookup in less than a second.

If I use Edge (details as per original entry), it can take 5-10 seconds before the results are returned.

magicbug, 2023-10-17:
Thats helpful so this is an edge issue I'll see if I can dig into it not sure why edge would cause such a problem

zaubererty, 2023-10-20:
I had the same issue on Chrome, when I updated form 2.4.5 to 2.4.10
Reason for the issue where a lot of radio requests that caused the tab to freeze. I solved that by deleting the cookies and the cache of the browser

closed 2023-10-24 by magicbug