magicbug/Cloudlog #2610: Feature Request: Parity with hamlib modes

ctand, 2023-10-22:
Cloudlog should maintain parity with all of the modes that hamlib supports for compatibility with third party rig control programs.

USB, LSB, CW, CWR, RTTY, RTTYR, AM, FM, WFM, AMS, PKTLSB, PKTUSB, PKTFM, ECSSUSB, ECSSLSB, FAX, SAM, SAL, SAH, DSB

Thanks!

magicbug, 2023-10-23:
Cloudlogs mode support is in line with the ADIF Specification, so there's little reason for it to support how rig control applications handle it, you can of course define modes yourself in the Admin area of Cloudlog but please remember if your moving logs between third party services this might give you problems.

closed 2023-10-23 by magicbug

ctand, 2023-10-23:
If this is the case then CloudLog should handle mismatched modes more gracefully. Currently when an undefined (but entirely accurate) mode is passed from radio control software to CloudLog, it defaults to NULL and CloudLog sets the propagation mode to Satellite. It should not set the propagation mode automatically on an error.