magicbug/Cloudlog #2611: Any Radio API update blanks all potential updatable fields

ctand, 2023-10-23:
When one makes an API call to update a single field such as frequency while in Live QSO mode, all potential fields covered by the API are updated as NULL if not set by the API call for some strange reason. This is problematic if you wanted to manually set the mode and have only the frequency updated via the Radio API. It is currently impossible to log anything with CAT updates unless it is very simple where the ADIF spec matches the mode reported by the radio such as CW or FM.

If there can't be some smarter way to interpret hamlib supported modes that are not ADIF compatible, there at least has to be a way to manually set these fields without them getting clobbered.

int2001, 2023-11-09:
What Client do you use?
At the first look this problem has to be solved at the Client which produces the API-Calls.

Indeed some Radios and Libs report e.g. USB-D or DIGI-U instead of USB. But that translation needs to be done at the Client which calls the CL-API.

Maybe i didn't understand the issue correctly. So if you have more Informations, please share....

ctand, 2023-11-11:
I'm using the API directly from my own script. You can't update only one field without all possible updatable fields updating to null if they aren't set by the call. CloudLog should only update fields directly updated by the API.

If the fields that aren't being called aren't nulled, then a user could manually set modes that the radio and software cannot report to CloudLog in the format you'd like it to but still have the frequency reported correctly.

What I'm asking for is this, only update the explicit field called by the API. If I make a call to update only the frequency, it  should only update the frequency. That call shouldn't blank a manual setting for mode just because I didn't set it in the API call.

int2001, 2023-11-14:
@ctand Does this fit your needs?

ctand, 2023-11-14:
Thank you! That works as intended. I'm on master so I applied it as a patch. I did however get these whitespace errors.

`2684.patch:49: space before tab in indent.
                                        if (isset($mode) && ($mode != null)) { $a_ret['mode']=$mode; }
2684.patch:87: space before tab in indent.
                                                old_mode = $(".mode").val();
2684.patch:88: space before tab in indent.
                                                $(".mode").val(data.mode);
2684.patch:90: space before tab in indent.
                                                old_mode = $(".mode").val();
warning: 4 lines add whitespace errors.`

int2001, 2023-11-14:
oh, i think the patch hasn't patched the JS correctly. You patched master, right?
That patch is based upon dev.
Maybe you wait until next master-release?

But tnx for cnfm

ctand, 2023-11-14:
Yep no problem. I unpatched and will wait for the next master. Thanks!

commit 0325d5c8 (2023-11-13): Merge pull request #2684 from int2001/singlecat

Enable QRG-Only-Update (should fix #2611)
commit cce52c12 (2023-11-13): Enable QRG-Only-Update (should fix #2611)
closed 2023-11-14 by magicbug