magicbug/Cloudlog #2623: Contest logging: Worked station do not appear on list.

205er, 2023-10-25:
Hi,

I use Cloudlog for all my Ham radio activities, including contest logging. And I am very pleased with Cloudlog! Thanks for your effort and good work!
Unfortunately, I discovered a minor glitch, when logging contest QSO:
Already worked stations appear on the bottom list, as soon as I enter a call sign. This only works, when I open "live contest logging" freshly. All stations that have been entered during a "session" (without accessing this page again through the menu) will never appear in this list. Nevertheless, the warning message "Worked before!" will be displayed, even if the QSO entry in the list is missing. This is somehow confusing. The attached screenshot will show the issue:
[image]
Is this a bug or just a issue of my server configuration?

Thanks & br!

dk5cr, 2023-10-30:
Same issue with my version. Was the same before updateing to 2.4.11 (2.4.5 before). Also the "Worked before" sign with white letters in red box doesn't appear when entering a Callsigh, which is worked before. Both is SSB and 80m, but no "match". The worked station (M6T) is visible in the list of all QSO and it's in the same contest. 
Problem occurs everytime you switch to another menu item in CloudLog and then switch back to (live) contest logging. Contest is - of course - correctly selected.

205er, 2023-10-31:
The "Worked before" warning appears as soon as you leave the call sign field (switch to next input field). It might take a moment, but that always works for me.
It's only the non-appearance of QSO in the Contest logbook (on the bottom), that bothers me.

dk5cr, 2023-10-31:
Thanks for the reply, but after typing in the callsign and leaving the field, drinkin' a beer nothing appears like "worked before". 
Perhaps it will be shown, if the worked callsign is in the list under the entry fields but... 

So I hope for a quick solution to this issue

phl0, 2023-11-09:
Hej @205er can you please have a look the JS console of your browser if any errors are reported?

205er, 2023-11-12:
Hi @phl0,

I just checked the contest logging fuctionality again and found the following errors in my browser's dev tool:
[image]
[image]

I am using Cloudlog 2.5.0.

phl0, 2023-11-12:
https://github.com/magicbug/Cloudlog/pull/2670 should fix this.

HB9HIL, 2023-11-12:
> #2670 should fix this.

@205er Will be released soon.

dk5cr, 2023-11-14:
Hi, 
just checked again: When i log a callsign which is already logged, the hint "Worked before" doesn't appear. 
Update to 2.5.0 is up to date. 
Fix lookups https://github.com/magicbug/Cloudlog/pull/2670 is done. 

One more question to this: How long does a contest sessions last? If i have a contest for few days, weekend or so, then does the session last for both days? What if there's a server restart (daily restart to cleanup the system)? 
Or is it just a new sessions after starting a new one with click on "New Contest session" button (perhaps the English name variies, German name is "Beginne neue Contest-Sitzung") . 

Thanks

closed 2023-11-17 by 205er