magicbug/Cloudlog #2628: Tab URLs in DXcluster Bandmap pointing to wrong targets

dm7ds, 2023-10-26:
If you want to switch the view in DXCluster [BandMap | BandList] you get a 404.

Links in
application/views/bandmap/index.php
application/views/bandmap/list.php

are set wrong.

```
diff --git a/application/views/bandmap/index.php b/application/views/bandmap/index.php
index f5c353b7..20f8af24 100644
--- a/application/views/bandmap/index.php
+++ b/application/views/bandmap/index.php
@@ -12,10 +12,10 @@
 <div class="tabs" id="dxtabs">
                <ul class="nav nav-tabs" id="myTab" role="tablist">
                        <li class="nav-item">
-                               <a class="nav-link active" href="index">BandMap</a>
+                               <a class="nav-link active" href="bandmap">BandMap</a>
                        </li>
                        <li class="nav-item">
-                               <a class="nav-link" href="list">BandList</a>
+                               <a class="nav-link" href="bandmap/list">BandList</a>
                        </li>
                </ul>
 </div>
diff --git a/application/views/bandmap/list.php b/application/views/bandmap/list.php
index 9b24f2d9..75d1eb60 100644
--- a/application/views/bandmap/list.php
+++ b/application/views/bandmap/list.php
@@ -31,10 +31,10 @@
 <div id="dxtabs" class="tabs">
                <ul class="nav nav-tabs" id="myTab" role="tablist">
                        <li class="nav-item">
-                               <a class="nav-link" href="index">BandMap</a>
+                               <a class="nav-link" href="bandmap">BandMap</a>
                        </li>
                        <li class="nav-item">
-                               <a class="nav-link active" href="list">BandList</a>
+                               <a class="nav-link active" href="bandmap/list">BandList</a>
                        </li>
                </ul>
 </div>
```

After correcting it this way, everything works fine.

magicbug, 2023-11-01:
@int2001 is this something we need to action into the codebase?

int2001, 2023-11-01:
I am not really sure. If the fact that "Bandmap" from the Dropdown-Menu links to the List is meant: that was intended, because the List is more verbose and default.

If that's not the issue, please describe. Because: if Bandlist (or map) is active the links are okay. Can't figure out a problem there.

magicbug, 2023-11-01:
Just tried it here too and I can jump between the tabs without any issue

dm7ds, 2023-11-01:
Dang! Sorry, I'm in the dev branch. Sorry, forgot to mention this or think about.

closed 2023-11-01 by magicbug