magicbug/Cloudlog #2633: Uncaught mysqli_sql_exception when trying to install

Eric-Sparks, 2023-10-28:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When attempting to install Cloudlog I'm receiving the following message:

`Fatal error: Uncaught mysqli_sql_exception: No such file or directory in /usr/local/www/apache24/data/install/includes/database_class.php:9 Stack trace: #0 /usr/local/www/apache24/data/install/includes/database_class.php(9): mysqli->__construct('localhost', 'wg3k', 'wx#GqXVxMVG2%zw...', '') #1 /usr/local/www/apache24/data/install/index.php(49): Database->create_database(Array) #2 {main} thrown in /usr/local/www/apache24/data/install/includes/database_class.php on line 9`

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '<url>/install'
2. Click on 'Install' once the fields have been added.
3. See error

**Expected behaviour**
I expect the software to go through the install process.

**Desktop (please complete the following information):**
 - OS: Fedora Linux 38
 - Browser: Firefox
 - Version: 119.0

** Server (please complete the following information):**
 - OS: FreeBSD 13

**Additional context**
Running Apache 2.4, PHP 8.1.24, and mariadb 10.11.  I have confirmed the credentials provided are correct and that php81-mysqli-8.1.24 is installed.

HB9HIL, 2023-10-31:
Cloudlog is developed on Debian/Ubuntu and uses the Codeigniter Framework. I'm not sure about depencies for Cloudlog on FreeBSD.

magicbug, 2023-10-31:
Got to say I'm not 100% sure I can help but reading the error message is the database created?

Beyond that I'm at a loss but this is an issue between PHP and MySQL.

HB9HIL, 2023-10-31:
@Eric-Sparks I was able to reproduce this error with FreeBSD and PHP 8.2. It's partially the same error message. 

As @magicbug mentioned it seems like you didn't create the database, did you?

Create a Database-User and a Database and try again.

magicbug, 2023-11-01:
I'm planning to close this issue, if you need any help with this issue I'd suggest using the discussion section

closed 2023-11-01 by magicbug

Eric-Sparks, 2023-11-01:
> @Eric-Sparks I was able to reproduce this error with FreeBSD and PHP 8.2. It's partially the same error message.
> 
> As @magicbug mentioned it seems like you didn't create the database, did you?
> 
> Create a Database-User and a Database and try again.

Yes, the database and the database user were created prior to running the install script.  I can log into mariadb (mysql -u <user> -p) using the credentials I am giving to the install script.

Eric-Sparks, 2023-11-01:
I've moved this to [discussion 2644](https://github.com/magicbug/Cloudlog/discussions/2644).