magicbug/Cloudlog #2646: Station Location County not pushed to QRZ

ae5au, 2023-11-02:
QSOs pushed to QRZ don't include the proper US county. They default to the home county on QRZ.

Station City, Station State, and Station Gridsquare are uploaded based on the station location from the QSO, but Station County is ignored.

Steps to reproduce the behaviour:
1. Create a new QSO with a specific station location that has fields set.
2. Push to QRZ (tested with same results using real-time and batched push).
3. Compare record on QRZ to record from station location.

Station location:
[image]

Record on QRZ:
[image]

ae5au, 2023-11-02:
I also verified that the same QSO does send the county to LoTW.

[image]

Also, I know that manually uploading an ADIF file to QRZ with MY_CNTY set will populate the county properly as this is how I've been handing my POTA activations for a while now.

magicbug, 2023-11-02:
thanks for reporting that does indeed look like a bug!

AndreasK79, 2023-11-04:
The problem lies in the component that generates our ADIF file. Since the county should be in the format of state, county.

```
<Two-letter state abbr.>
 
COMMA
 
<county name>
```
We only output the station county in my_cnty, so we need an update here.

AndreasK79, 2023-11-04:
I pushed a fix to the dev branch that should take care of this.

closed 2023-11-10 by magicbug