magicbug/Cloudlog #2666: logbookadvanced page table footnote is incorrect

littlebear0729, 2023-11-08:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
I have 637 QSOs.
When the page size is 250, the footnote below the table shows 'Showing 1 to 250 of 250 entries', and it should be 'Showing 1 to 250 of 637 entries'
When the page size is 1000 or more, it shows 'Showing 1 to 637 of 637 entries' which is correct.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to advanced logbook page
2. Click on Search Button
3. Change page size
4. Click on Search Button

**Expected behaviour**
I have 637 QSOs.
When the page size is 250, the footnote below the table shows 'Showing 1 to 250 of 250 entries', and it should be 'Showing 1 to 250 of 637 entries'
When the page size is 1000 or more, it shows 'Showing 1 to 637 of 637 entries' which is correct.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - Version 2.5.0

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

abarrau, 2023-11-08:
Hello, 
in fact, on logbookadvanced page, table return the number defined (and limit result with this number). 
you can see that there is no pagination offered.
To see how the cloudlog team wishes to implement this results table ?

HB9HIL, 2023-11-08:
@littlebear0729 

I understand your expectation to see `Showing 1 to 250 of 637 entries`
But I think there is a difference between expectation and what this "#Results Dropdown" does. I'll try to explain. 

This is no bug, rather it does exactly how it is designed. 

[image]

This Dropdown acts like a filter. 
Maybe your expectation was `Show me max. 250 QSO's of all my QSO's in Logbook`

But it acts like `Show me max. 250 QSO's in this list`

So if you need a number of all QSO's you can use for example the Analytics > Statistics Page on Tab "QSO" 

I hope I was able to explain why there is a difference between User Expectation and Design of the Logbook Advanced

closed 2023-11-08 by magicbug