magicbug/Cloudlog #2689: [Feature Request] Configuration option for CSV separator

m0urs, 2023-11-14:
It would be great to have possibility to configure the separator character for all CSV exports. Currently the character ist fixed to a "," (comma). Unfortunately SM Excel is not able to open the CSV file correctly if it is configured to German language. In this case a ";" (semicolon) is required as separator.

Although you can download the CSV file, change it with a text editor and then open it in Excel, it would be more convenient if you just could configure the character somewhere in the settings.

phl0, 2023-11-14:
I guess this is not needed. You can tell Excel upon opening a csv file which char is used to separate fields. This character is *not* depending on language (which would not make any sense imho).

m0urs, 2023-11-14:
Unfortunately it IS depending on your country settings :-(

"To handle .csv files, Microsoft Excel uses the List separator defined in [Windows Regional settings](https://www.ablebits.com/office-addins-blog/change-excel-csv-delimiter/#default)."

And you cannot specify the character while double clicking on an CSV file. If the CSV file has the wrong separator, you need to open a blank Excel sheet and then importing the CSV file via the CSV import wizard.... 

Yes, you can do that or you can change the file with a text editor.

However it would just be far more easier to just be able to specify the character in the settings :-)

phl0, 2023-11-14:
I guess the "Test to columns" function in Excel can help you after opening the csv file. Also allows for setting delimiting quotes etc.

[image]

m0urs, 2023-11-14:
Ok, yes, that is quite similar to the CSV Import wizard. I then will do it that way.

closed 2023-11-14 by m0urs