magicbug/Cloudlog #2697: Docker: Updates do not cleanly happen with documented docker-compose configuration

bgritzmacher, 2023-11-15:
When using the docker-compose as specified in the project's wiki, migration updates will never run automatically after pulling the latest image, resulting in an outdated database schema.

This is due to the entire `application/config` directory being mounted to a persistent docker volume, which gets laid over the container's file system upon being mounted by Docker.

This means the `migration.php` file gets overwritten with the old version of it that exists within the volume, so it never knows to run the new migrations. This also means that any changes to files within the config directory never apply on update, either.

I can work around this by running `git restore` on changed files to update the volume file system, however, expected behavior should be that all one needs to do for an update with docker is to pull the new image and restart the cloudlog service with the new image.

Ideally, I would only store what needs to be persistent in the `cloudlog-config` volume - which is just `config.php` and `database.php`, but due to how Docker does volume mounts, the file *has* to exist within image first before applying the volume layer, otherwise it gets created as a directory.

The other possibility is to use filesystem bind mounts, but:

- It is my preference in Docker to avoid doing this for "production" and not have unnecessary files sitting around in the file system.
- I believe there are some inconsistencies in how bind mounts are handled between Linux and Windows systems (up to and including it not being possible at all.)
- A Docker image should be flexible enough to handle the file being in either the image, a bind mount, or in a Docker volume.

Letting the config directory live and die with the container and just accept going through the install every time an update happens, while not ideal, is also not acceptable because it appears at present, the `install.sql` script does `DROP TABLE IF EXISTS` on many tables, which results in lost data.

rayslinky, 2023-11-16:
Similar to https://github.com/magicbug/Cloudlog/issues/2485

iu2frl, 2023-11-26:
@magicbug this is a very detailed analysis and i think this explains #2308 which happened in the same way today, i think the critical point is that the install script should be somewhere outside `var/www/html/application/config` as this is only run once, or to store the config files somewhere else. I have no idea if this is supported by the current framework though.

magicbug, 2023-12-12:
Lets be clear as a project I have decided not to provide docker support going forward, as this isn't code related if you need help with setting up server components use discussions

closed 2023-12-12 by magicbug

iu2frl, 2023-12-12:
You are definitely right but this is not docker only, with a manual update the user would still need to align the `config.php` file

HB9HIL, 2023-12-12:
> You are definitely right but this is not docker only, with a manual update the user would still need to align the `config.php` file

config.php is catched by .gitignore

if a user updates cloudlog with the update script (which uses git) the config.php wont be touched

iu2frl, 2023-12-12:
And then you got stuck in my situation where the translation array in that file got updated and some pages were not showing any text (such as the Adif import or the advanced qso search)