magicbug/Cloudlog #2703: Start Date/Time gets deleted when editing existing QSO

205er, 2023-11-17:
Hi all,

I found a bug, that is very annoying. I logged a QSO and afterwards I need to make some changes to it, so I edit the QSO and save my corrections. Unfortunately, the start date/time will then be set automatically to _0000-00-00 00:00:00_ (while the end date stay untouched)
[image]

[image]

This issue will only appear when changing an existing QSO (at the right side) from "Live QSO"

magicbug, 2023-11-20:
Yep seems resolved now in dev branch

closed 2023-11-20 by magicbug

phl0, 2023-11-20:
I can confirm that this fixes/fixed the bug.

205er, 2023-11-21:
Thanks HB9HIL for you fix! Hope to meet you on the bands soon! ..und 73 von der anderen Bodenseeseite.

PR #2710 (2023-11-20): Fix QSO Start Time gets overwritten when editing a QSO in Live QSO Logging
This should fix #2703 

Please test and confirm. Tested different QSO Edits and everything seems still working after removing those classes from the Form Input.
commit c2409ec5 (2023-11-20): removed input classes to fix  #2703