magicbug/Cloudlog #2704: Comments will get (partially) deleted when including quotation marks

205er, 2023-11-17:
Hi,

today I discovered that comments will get (partially) deleted after saving when the text including quotation marks. Here is an example:

First I enter the comment...
[image]
...and save the QSO data set.

The comment is visible in viewing mode
[image]

But after reopening the QSO, the comment text partially disappeared.
[image]

There seems to be a function that interprets quotation marks incorrectly.

HB9HIL, 2023-11-17:
@205er Thx for reporting this bug. :)

205er, 2023-11-17:
*wow* that was a quick bug fix! Thanks for your help!

HB9HIL, 2023-11-17:
have to get proofed by the others but here this works. was able to reproduce that issue :)

PR #2705 (2023-11-18): Fix comment quotation marks
This should fix #2704 

Please check and test
closed 2023-11-20 by magicbug