magicbug/Cloudlog #2706: Advanced Log: Search Dupes only worked for one time

dg9vh, 2023-11-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
The button "Dupes" in "Advanced Log" only worked for exactly one time and now produces error-popup.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Logbook - advanced
2. Click on Dupes-Button
3. remove dupes by deleting them (the first found ones in the result-set of 250 logentries)
4. Do it a second time
5. from now on see the following error:
[image]

**Expected behaviour**
It should present a new list of dupes found. It seems, that it does not found any dupes and therefore throws this error.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Desktop (please complete the following information):**
 - OS: ChromeOS, but error is also reproduceable on Windows and Linux
 - Browser Chrome
 - Version 120.0.6099.25 

**Additional context**
If I should deliver more information, let me know.

Here are some lines from application/log

`
ERROR - 2023-11-17 17:00:57 --> 404 Page Not Found: Assets/js
ERROR - 2023-11-17 17:01:11 --> Severity: error --> Exception: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ')

                        ORDER BY qsos.COL_TIME_ON desc, qsos.COL_PRIMARY_KEY desc
                        LIMIT 250' at line 12 /home/huebelon/public_html/dg9vh.de/cloudlog/system/database/drivers/mysqli/mysqli_driver.php 301
ERROR - 2023-11-17 17:01:12 --> Severity: error --> Exception: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ')

                        ORDER BY qsos.COL_TIME_ON desc, qsos.COL_PRIMARY_KEY desc
                        LIMIT 250' at line 12 /home/huebelon/public_html/dg9vh.de/cloudlog/system/database/drivers/mysqli/mysqli_driver.php 301
`

int2001, 2023-11-18:
can you specify the exact CL-Version (taken from admin / debug-info)?

and: is there any kind of ADIF, since we were not able to reproduce this one. i think there must be a specific situation with dupes at DB?

int2001, 2023-11-23:
@dg9vh - no chance to reproduce the issue. What should we do?!

dg9vh, 2023-11-23:
> @dg9vh - no chance to reproduce the issue. What should we do?!

I debugged it today with @phl0 and we found the issue, that it happens because of >4000 dupes in the database... he knows details about it. He also could reproduce it on his dev-setup so I think there could be a solution soon.

Actually we hotfixed it by modifying the list of IDs created and removing a tailing "," in the list...

Just waiting for final.

phl0, 2023-11-23:
To be honest my hotfix does not solve the problem. The issue is with group_concat function of SQL. It is limited to 1024 chars. Removing the trailing comma is only half a solution. Actually the list might get cut in the middle of a QSO number resulting in a non-dupe QSO being listed. As this should be noticable by the user it should be safe. But still trying to find a better solution.

AndreasK79, 2023-11-24:
@phl0's fix is merged. Excellent work digging into this and finding the reason.

@dg9vh thanks for reporting this. This was a tough one. Closing this, as dev contains the fix now.

closed 2023-11-24 by AndreasK79