magicbug/Cloudlog #2712: SSL Database can't connect

BloodBladeW, 2023-11-21:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
My database provider(PlanetScale) forcibly uses SSL/TLS to connect to the database. After modifying the database.sample.php file, still prompt that I can't connect.

**To Reproduce**
Steps to reproduce the behaviour:

1. Edit application/config/database.sample.php
```
        'encrypt' => TRUE,
        'ssl_ca' => '/etc/ssl/certs/ca-certificates.crt',
        'ssl_verify' => TRUE,
```

2. Visit the installation page and fill in the database information.

3.Prompt error:
```
Fatal error: Uncaught mysqli_sql_exception: unknown error: Code: UNAVAILABLE server does not allow insecure connections, client must use SSL/TLS in /var/www/html/install/includes/database_class.php:9 Stack trace: #0 /var/www/html/install/includes/database_class.php(9): mysqli->__construct('<delete database host and username>, Object(SensitiveParameterValue), '') #1 /var/www/html/install/index.php(49): Database->create_database(Array) #2 {main} thrown in /var/www/html/install/includes/database_class.php on line 9
```

**Expected behaviour**
Able to connect to the database and use it.

In addition, it is hoped that the option of using SSL connection can be added to the installation page.

**Desktop (please complete the following information):**
 - OS: Use Docker on Debian

**Additional context**
I also noticed that database providers say that they do not support foreign key constraints because they use MySQL clusters. Will Cloudlog use foreign key constraints? If Cloudlog use it, I may look for a database provider that does not use clusters.

magicbug, 2023-11-21:
I think you're not correctly setting up the array and it would need to be along the lines of 

```php
'encrypt' => [
        'ssl_ca' => '/etc/ssl/certs/ca-certificates.crt',
        'ssl_verify' => TRUE,
],
```

magicbug, 2023-11-21:
As this isn't a bug with Cloudlog I'm going to convert to a discussion