magicbug/Cloudlog #2714: VUCC award worked/confirmed gives bogus values

pa8s, 2023-11-21:
**Describe the bug**
On the main page the VUCC stats gives bogus values.

I have no idea where the worked (991) and confirmed (886) values come from. The only VUCC band in my log is 6m with only 12 qsos of which 10 are confirmed. 

**To Reproduce**
n/a

**Expected behaviour**
VUCC values representing logged QSOs

**Server**
OS: FreeBSD 13.1
Server: nginx/1.24.0
PHP: 8.1.25
Git branch: dev
Commit: https://github.com/magicbug/Cloudlog/commit/694f9fbd80cb7ce28286efbc0d280e6f238f5c0a
Tag: https://github.com/magicbug/Cloudlog/releases/tag/2.5.0

int2001, 2023-11-23:
@pa8s i think we can close this now?!

pa8s, 2023-11-23:
Issue solved. Thank you

PR #2716 (2023-11-22): Splitted VUCC into real VUCC and SAT-VUCC at Dashboard
Should fix #2714 

Old behaviour:
- Every grid was counted at Dashboard (ignore of Band / Mode)

New behaviour:
- Only >6m is counted. It's splitted into two columns:
- All QSOs >6m (except SAT)
- Only SAT

[image]
closed 2023-11-23 by pa8s