magicbug/Cloudlog #2721: QSOs entered via the Simple Fast Log Entry screen have an invalid frequency

TeknikalDomain, 2023-11-24:
**Describe the bug**
When a QSO is entered via the SFLE page, frequencies saved are nonsensical, such as "14" instead of the more appropriate "14000000" (14MHz).

**To Reproduce**
Steps to reproduce the behaviour:
1. Enter the Simple Fast Log Entry page from QSO > Simple Fast Log Entry
2. Fill out the date and station information
3. Enter a QSO, such as `1234 7.233 w2taz`
4. Press "Save in Cloudlog" and confirm
5. Open the newly-added QSO and press Edit QSO
6. The resulting frequency will be `7` instead of the expected `7233000`.

Alternatively for step 3, specify the QRG beforehand, such as in this example:
```
40m 7.233 ssb
1234 w2taz
```

**Expected behaviour**
When a frequency is entered via the SFLE page, exact frequencies should be saved in the QSO records.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Multiple (server-side issue affecting multiple devices)
 - Browser: Multiple (server-side issue affecting multiple devices)
 - Version: Multiple (server-side issue affecting multiple devices)

**Additional context**
According to the Fast Log Entry documentation, the correct way to specify a QRG is to add a number with _a decimal point._ Thus, `7.123` is a QRG, but `7.235.000`, `7,123,000`, or `7.233mhz` are **not.** It seems somehow this value is being taken as the value in _Hz_ and not as it should be interpreted, in _MHz,_ which is causing it to get truncated (from 7.222 to just 7). I tried to test this by using `1234 7233000 w2taz` as a QSO line, but that saved with a frequency of 0 instead.

HB9HIL, 2023-11-24:
Please update your Cloudlog Instance to the latest version.

HB9HIL, 2023-11-24:
This should already be fixed but it seems like you are using to the latest version of Cloudlog.

HB9HIL, 2023-11-24:
After you updated the correct syntax for this qso qould be
```
ssb
1234 7.233 w2taz
```
as you described in point 3. 👍  The date you can choose in the date field or as 
```
date 2023-11-24
ssb
1234 7.233 w2taz
```

TeknikalDomain, 2023-11-24:
Should have been updated fairly recently, give me a second to check.

HB9HIL, 2023-11-24:
> Should have been updated fairly recently, give me a second to check.

Set a cronjob to update your Cloudlog Instance every night 👍  😉

TeknikalDomain, 2023-11-24:
Freshly updated, from 2.5.0 to 2.5.2. Confirmed, fix was in one of those two patch versions.

closed 2023-11-24 by TeknikalDomain

HB9HIL, 2023-11-24:
> Freshly updated, from 2.5.0 to 2.5.2. Confirmed, fix was in one of those two patch versions.

I fixed this in 2.5.1 and I saw you are using 2.5.0 because the UTC time stamp came with 2.5.1 👍 

Thx for using Cloudlog and reporting Problems 

We really appreciate that :)

TeknikalDomain, 2023-11-24:
Looks like my other side-gripe that it didn't support "lsb" or "usb" for SSB sub-modes also got resolved somewhere in there, neat.

(Yes, I'm overly pedantic with my logging)

HB9HIL, 2023-11-24:
> Looks like my other side-gripe that it didn't support "lsb" or "usb" for SSB sub-modes also got resolved somewhere in there, neat.
> 
> (Yes, I'm overly pedantic with my logging)

Absolutely not pedantic :) yep got also fixed. Let us know if you find some more!

HB9HIL, 2023-11-24:
It also supports the `day +`or `day +++....` command now. 

You can see that in the Syntax Help.

TeknikalDomain, 2023-11-24:
Oh there's a few other minor things. The "bands" list not having the 630m (472 kHz) or 2200m (135kHz) MF / LF bands, the live / post QSO screens not populating name / location / grid when you enter a call (which can be done with HamDB if I recall correctly, or a QRZ subscription), VUCC counting all QSOs, not just VHF / UHF, or that the "Verified (Match)" settings for QSL confirmations don't actually update the display (the "received" arrow still shows red as if it's not been confirmed), but those are all minor enough that I don't really know if it's worth opening issues for (except maybe the last one, that might be relevant).

Honestly great work.

HB9HIL, 2023-11-24:
> Oh there's a few other minor things. The "bands" list not having the 630m (472 kHz) or 2200m (135kHz) MF / LF bands, the live / post QSO screens not populating name / location / grid when you enter [...]

No Problem :) 

- In the "Bands" List you can create new Bands as you need them. Not that much people are using 630 and 2200 so they are not added by default. If you need them you can create them. I can add support for these two bands on simplefle if this is needed. 
- the Live/post QSO problem you could describe a bit more precise or open an issue for that.
- VUCC display was just fixed on the dashboard in 2.5.2. For the VUCC awards page I would recommend to disable the not needed bands in Bands List. You can define with the checkboxes where the bands should be used and where not.
- The verified mark works here as expected. You could describe that more precise and open an issue to find out if there is a bug or just an config problem. 

:)