magicbug/Cloudlog #2727: mysqli_sql_exception error

ofadam, 2023-11-24:
**Describe the bug**
On two different installations on the same server, I'm suddenly getting this PHP error despite not having changed anything intentionally:

An uncaught Exception was encountered

Type: mysqli_sql_exception

Message: All parts of a PRIMARY KEY must be NOT NULL; if you need NULL in a key, use UNIQUE instead

Filename: /var/www/kd9nye/html/system/database/drivers/mysqli/mysqli_driver.php

Line Number: 301

**Desktop (please complete the following information):**
 - OS: MacOS 14.1.1
 - Browser: Firefox 119

**Server info**
- Ubuntu: 22.04.3
- PHP: 8.3
- Apache: 2.4.54
- MySQL: 8.0.35-0ubuntu0.22.04.1

phl0, 2023-11-24:
This known and corrected by a fix committed earlier today. Just pull dev again.

ofadam, 2023-11-24:
Sure enough – I'm back in business! Thanks much.

closed 2023-11-24 by ofadam