magicbug/Cloudlog #2736: Lotw Upload / Livestream Clublog cron job not working

GregTroar, 2023-11-27:
Me again sorry :(

I use last dev, windows 11, server is on Debian Bookworm.

I have set my cronjob as per the manual https://url-to-cloudlog/index.php/lotw/lotw_upload

And I get the rror:

"You have not defined your ARRL LoTW crednetials"

But I definitely have and I am sure they are correct.

If I paste in chrome the link this is what I get

[image]

If I use the link in the error directly in chrome it works just fine.

For Clublog I used the cronjob as in the wiki but that does not work. If I just do a curl https://log.xxxx.com/index.php/clublog/realtme/cloudloglogin and hit enter it is processed instantly without any error.

If I use same link in chrome directly it works.

GregTroar, 2023-11-27:
Ok so nothing I can do, any hint about why curl command for clublog realtime does not work ?!
Dunno where to look to find the issue.

phl0, 2023-11-27:
I am just testing. My request in the browser is running for minutes now. I guess curl simply runs into a timeout. Digging further.

GregTroar, 2023-11-27:
Mine for clublog is instant and nothing happens.

For lotw I tried again the url directly in Chrome and it works but from the cronjob it still says wrong credentials while URL is exactly the same I use https://log.xxxx.com/index.php/lotw/lotw_upload

Works in chrome does not in curl on same server as cloudlog...sounds very strange. Will be a pain to do this all manually especially for realtime clublog.

phl0, 2023-11-27:
Just a note on my way debugging this: curl fails because it does not have a valid session with Cloudlog and the code checks for that. So this will simply return no profiles to upload QSOs for.

GregTroar, 2023-11-27:
And how can I solve this ? (get curl to have a valid session?)
Thanks

phl0, 2023-11-27:
No no. We are working on a solution. @int2001 will come up with a fix shortly.

GregTroar, 2023-11-27:
Oh ok nice !
Thanks a lot

int2001, 2023-11-27:
> For Clublog I used the cronjob as in the wiki but that does not work. If I just do a curl https://log.xxxx.com/index.php/clublog/realtme/cloudloglogin and hit enter it is processed instantly without any error.
> 
> If I use same link in chrome directly it works.

the correct link for Clublog-Upload via cron is: http://[your instance]/index.php/clublog/upload
not the realtime thing. can u please try that - that should work, when uploading via cron.

Currently there's no RT-Support for Clublog. But - as @phl0 told - i'm investigating how to make it possible.

GregTroar, 2023-11-27:
This one works without any issue but what I am looking for in the cron job is the realtime upload to have the clublog livestream being updated live.

So what I am looking for is the https://log.xxxx.com/index.php/clublog/realtme/cloudloglogin cronjob to work.

Thanks

phl0, 2023-11-27:
Thats two different functions... I am not sure if the upload function also feeds data for Clublog realtime.

HB9HIL, 2023-11-27:
> Thats two different functions... I am not sure if the upload function also feeds data for Clublog realtime.

I set the normal upload function to upload all 5 min :D thats okay for me

phl0, 2023-11-27:
Does this data show up in clublog reatime feed?

int2001, 2023-11-27:
yes. understood. both jobs are doing nearly the same. they're using different API-Endpoints at clublog, but goal is equal: upload the qsos to clublog.

that "realtime"-cron isn't really realtime. it is called every minute or every 2 minutes. Realtime will be "instant" after the QSO, as it is implemented for qrz.com.

i try to check if this is possible for Clublog, too - without a cron

HB9HIL, 2023-11-27:
> Does this data show up in clublog reatime feed?

yes.. why should it not?

the clublog realtime view get his data from the normal clublog data.

GregTroar, 2023-11-27:
Yes it will show in realtime feed if normal upload.
Honestly this is very convenient to have the livestream especially when running a pile up here in Vietnam, 5 minutes is actually quite long, people likes to see if they are in the log just after the contact or will call again within minutes and this will create a dupe for us...

I could probably use the upload function every minute but not sure this is good on clublog side as I believe they have a specific API for realtime upload

HB9HIL, 2023-11-27:
we are talking abot this one https://clublog.org/livestream/hb9hil/m

GregTroar, 2023-11-27:
yes exactly https://clublog.org/livestream/xv9q

phl0, 2023-11-27:
> yes.. why should it not?

I was wondering why there is a separate function in our code then. If so, I would tend to add the realtime upload function to QSO saving code instead of calling it via cron. Just as qrz or DX Club uploads do...

GregTroar, 2023-11-27:
> the correct link for Clublog-Upload via cron is: http://[your instance]/index.php/clublog/upload not the realtime thing. can u please try that - that should work, when uploading via cron.
>
> Currently [...]

http://[your instance]/index.php/clublog/upload

This does not work either for me via cronjob

int2001, 2023-11-27:
> http://[your instance]/index.php/clublog/upload
> 
> This does not work either for me via cronjob

okay. any errors in the log? or what is happening? Did you create all of your station_profile-Callsigns at clublog?

GregTroar, 2023-11-27:
Yes of course if I use the link directly in Chrome it works fine.
The problem is exactly the same as with the realtime upload.
When I do the curl command it just instantly process it without any error, I believe this is the valid session issue of curl ?!?

int2001, 2023-11-27:
> Yes of course if I use the link directly in Chrome it works fine. The problem is exactly the same as with the realtime upload. When I do the curl command it just instantly process it without any error, I believe this is the valid session issue of curl ?!?

Ok, that's really strange. Neither http://[your instance]/index.php/clublog/upload nor /index.php/lotw/lotw_upload need a valid session. They're simply doing their job (silent!). If a qso exists it uploads it, and at LotW it'll also leech confirmations.

GregTroar, 2023-11-27:
ok let me re test all of it again

GregTroar, 2023-11-27:
So, the crontab job for lotw upload still does not work:

Upload Successful - ./uploads/lotw/xv9q-2023-11-27-15-24-33-cloudlog.tq8<br>You have not defined your ARRL LoTW credentials!

In chrome it works.

The Clublog normal upload works, only the realtime does not work.

int2001, 2023-11-27:
so, Cloudlog uploads one File/Callsign to LotW - at the other station or Call or User there are no LotW-Credentials. That's the reason for the message "You have not defined your ARRL LoTW credentials!".

Regarding Clublog - with the PR we are removing the endpoint for realtime. No more need to put it into crontab. The ClubLog Upload will happen automaticly after EVERY QSO without any cron.

GregTroar, 2023-11-30:
Hi,

I can only put one LoTW account under account.
It is my main lotw account under F4BPO callsign.

Inside this account on lotw I have several callsigns such as XV9Q or XV9BPO.

As I can only fill in Cloudlog my main account for LoTW this is what I did.
The cron upload for Lotw always return this no credentials error, not sure what I am supposed to do.
I cannot put credentials for XV9Q or XV9BPO anywhere in cloudlog unless I am missing something...

phl0, 2023-11-30:
The issue is there is a useless error message after all downloads have finished. I have fixed it in https://github.com/magicbug/Cloudlog/pull/2752. The download did work as expected it is just the error message is wrong. Most cron users will not have noted because as per Wiki then use `--silent` and discard the output.

phl0, 2023-11-30:
So after this you say that LoTW upload via cron is (still) not working for you?

GregTroar, 2023-11-30:
No I do not know actually.
I will do a test with a fake call to see if upload works.
I said it did not work just based on the error message !!!
Let me test again and I will let you know if the cron upload works.

GregTroar, 2023-12-01:
ok I do confirm that the LOTW upload works despite the error message which is supposed to be for downloading from Lotw.
So the cron works for upload.

Thanks

phl0, 2023-12-01:
Download was working as reported. So can we/you close this issue?

closed 2023-12-01 by GregTroar