magicbug/Cloudlog #2741: PHP error when trying to update missing grid

ofadam, 2023-11-28:
**Describe the bug**
When trying to update QSOs that are missing the gridsquare, I'm getting a PHP error.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to /update/check_missing_grid
2. A PHP error results for each QSO missing a grid

**Error Message**

A PHP Error was encountered

Severity: Warning

Message: Undefined array key "gridsquare"

Filename: models/Logbook_model.php

Line Number: 4061

**Server info**

    Cloudlog dev branch
    Ubuntu: 22.04.3
    PHP: 8.3
    Apache: 2.4.54
    MySQL: 8.0.35-0ubuntu0.22.04.1

phl0, 2023-11-29:
Fixed with https://github.com/magicbug/Cloudlog/pull/2745.

HB9HIL, 2023-11-29:
@ofadam 

Please check and close if the issue is gone. Fix is in 'dev' branch 😁

ofadam, 2023-11-29:
Hmm.. I'm now getting a long list of "Error: Callsign not found" errors when trying to update grids.

phl0, 2023-11-29:
> Hmm.. I'm now getting a long list of "Error: Callsign not found" errors when trying to update grids.

Yepp. Thats the case where the callsign does not have an entry in the callbook you are querying (qrz or hamqth).

ofadam, 2023-11-29:
Was that failing silently before recently? I don't remember seeing that error before. No biggie if just a visual change.

phl0, 2023-11-29:
The error has not been handled before. The code assumed that the answer always contained a grid which obviously is not the case. Now we are passing the error from the upstream callbook (e.g. qrz) to the user.

phl0, 2023-11-30:
@ofadam Can you confirm it is fixed? If so please close the issue. Tnx =)

closed 2023-12-07 by ofadam