magicbug/Cloudlog #2778: Advanced Logbook is empty

m0urs, 2023-12-05:
**Describe the bug**

In the current Dev the Advanced Logbook does no longer show any entries.

**To Reproduce**

Steps to reproduce the behaviour:

1. Go to 'Logbook -> Advanced'
2. You see that there are no QSOs listed

**Expected behaviour**

The last 250 QSOs should be displayed.

**Screenshots**

[image]

In the Master build everything still works fine.

phl0, 2023-12-05:
I cannot confirm. Could you please press CTRL + F5 and if that does not work please show the response/request after pressing the search button.

AndreasK79, 2023-12-05:
Can't reproduce here either. Could you go to options and save, then try?

m0urs, 2023-12-05:
Saving the options did repair it...! I only tried "Reset" before which did not change anything. Thanks a lot for your quick help!

PR #2782 (2023-12-06): Fix Empty LBA
Added POTA-Option, if pota isn't set.
Without that, LBA is empty and User has to call and save "options" once.

Resolves #2778
closed 2023-12-05 by m0urs