magicbug/Cloudlog #2794: CloudlogCat

ChrisL79, 2023-12-07:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Trying to install CloudlogCat on Windows 10 x64. On opening the program, there is an Axtive X error. Pressing continue opens the program. Then input the server address and api. The program will not connect to server.

**To Reproduce**
Steps to reproduce the behaviour:

1. Install CloudlogCat in W10 x64

**Expected behaviour**
For the program to connect to my instance. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

phl0, 2023-12-07:
Does anything appear in your Cloudlog log files?

ChrisL79, 2023-12-07:
Negative

ChrisL79, 2023-12-07:
> Have you installed Omnirig? do you see frequency info in cloudlogcat
> even with it not connecting to the server, can we see a screenshot of
> settings and the main window.  Remember to blank out your API key

Thanks for the reply. 

I will double check all of that tomorrow and post what i find. 

Thank you!

magicbug, 2023-12-12:
How did you get on?

ChrisL79, 2023-12-12:
Thanks for getting back to me.

I have Covid...AGAIN, which is making concentrating on things that bit harder.

I have had somewhat success with Aurora.

The issue i am having now is Aurora seems to be timing out if i do not make quick successive qsos. Any more than a minute without logging anything and its like Aurora goes to sleep. The only way to fix it is to restart Aurora.

Any idea?

Chris.