magicbug/Cloudlog #2804: LOTW cert import php error

arcreigh, 2023-12-09:
Running latest Docker build of cloudlog 2 PHP error messages are returned when uploading a lotw certificate.

A PHP Error was encountered
Severity: Warning

Message: Undefined array key "cert"

Filename: controllers/Lotw.php

Line Number: 407

and

A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /var/www/html/system/core/Exceptions.php:283)

Filename: helpers/url_helper.php

Line Number: 564

arcreigh, 2023-12-09:
I should also state no password is on this cert.

magicbug, 2023-12-12:
Lets be clear as a project I have decided not to provide docker support going forward, as this isn't code related if you need help with setting up server components use discussions

closed 2023-12-12 by magicbug

arcreigh, 2023-12-12:
What is the underlying issue so I can troubleshoot this. I don't mind helping out with docker support I just don't know PHP to be able to debug what the issue is.

HB9HIL, 2023-12-12:
@arcreigh As your docker wont get updates for the future I would recommend to make a proper backup and install it the traditional way as described in the Wiki. Then you try again uploading an LoTW Cert. If the issue persists we can check that. What the issue might be is hard to say.

arcreigh, 2023-12-12:
@HB9HIL A shame, docker is pretty useful and limit's erroneous system issues in my experience by having a standardized image. 

Is there a technical reason for dropping the support there? If there is i'd love to know what it is for my own learning purposes as my background is primarily network engineering. If it is more of a resource/timesync issue I'd be happy to help keep the docker side alive.

raufaser, 2024-01-19:
I get the same errors on a NON-docker installation.

```
A PHP Error was encountered
Severity: Warning
Message: Undefined array key "cert"
Filename: controllers/Lotw.php
Line Number: 407

A PHP Error was encountered
Severity: Warning
Message: Cannot modify header information - headers already sent by (output started at /var/www-cloudlog/system/core/Exceptions.php:283)
Filename: helpers/url_helper.php
Line Number: 564
```

Can this issue be reopened or shall i create a new one?

raufaser, 2024-01-20:
nvm i opened a new [issue](https://github.com/magicbug/Cloudlog/issues/2928)