magicbug/Cloudlog #2829: Local ADIF upload crashes server!

ChrisL79, 2023-12-13:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Importing local adif crashes cloudlog. Only possible to restart cloudlog by restaring apache.
In etc php, i have increased memory to 512mb, increased max upload file size to 5M, post size to 8M. Timeout set to 2 minutes. Setting timeout higher makes no difference.

Even when splitting the log to 2mb files it still crashes.

Error is 504. Pegs cpu at 100%!
.
I scanned my log for dupes beforehand, there were none l, although this should not matter.

Running ubuntu server 22.04 on i5 with 32gb.

What am i missing here?

**To Reproduce**
Steps to reproduce the behaviour:
1. Upload adif
2. 
3. 
4. 

**Expected behaviour**
For the adif log to upload without bringing the server to a crashing halt for everyone else, (or me)!
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

magicbug, 2023-12-13:
Sounds odd, for a machine thats got so much ram unless  the i5 isn't very "hot" but I'd suggest digging into server logs to see if anything is provided I don't believe we have a bug in the code.

This could also be MySQL running out of resources somehow, you can turn on logging in /application/config/config.php and then you'll find logs in /application/logs which might prove useful, apache logs might suggest something too.

504 is basically server timing out.

magicbug, 2023-12-13:
You might find increasing the Maximum PHP script run time helps but sounds like the server configs need some tweaking

ChrisL79, 2023-12-13:
Thank you. Plenty for to be getting on with there. Offhand, do you know locations for imcreasing mysql resources and php runtime?

HB9HIL, 2023-12-13:
@ChrisL79 

Please check the ADIF File with a validator as described on our Wiki Page:

[](https://github.com/magicbug/Cloudlog/wiki/ADIF-file-can't-be-imported)

HB9HIL, 2023-12-14:
@ChrisL79 were you able to test your adif file?

ChrisL79, 2023-12-14:
Yes. There were errors.

No matter how i export it from log4om there is errors.

phl0, 2023-12-14:
What does the CL logfile say about it? How may QSOs are in the file?

ChrisL79, 2023-12-14:
Around 7k. Tried all export options lile export as standard adif. Not got my head around log yet as suffering with Covid. Hard to concentrate.

phl0, 2023-12-14:
I can offer to try to import into my dev env if you want.

ChrisL79, 2023-12-14:
That is very kind. Thank you so much.  Let me know how i can get them to you and i will do it tomorrow.

phl0, 2023-12-14:
Find my email on qrz.com.

phl0, 2023-12-15:
File received and imported into my dev. Besides some dupes 4462 QSOs were imported without issues... Can you elaborate on your setup? Maybe script timeout on your webserver?

ChrisL79, 2023-12-15:
Thanks for the info.

Self hosted, dedicated server. i5-32gb
Script timeout possible. I need to tune settings. Do you have a few
examples of where to start with settings values? All i have changd so far
is max upload and post size.

How long did it take for you to import the log into your setup?

phl0, 2023-12-15:
The import took almost exactly 8 minutes.

ChrisL79, 2023-12-15:
Thank you. I need to find the reason it crashes my server.

phl0, 2023-12-15:
Check script timeout and the log files of your web server as well as Cloudlog itself.

ChrisL79, 2023-12-15:
I have logging set at 4 (everything) and nothing was logged at all regarding the error, or anything of interest. 
I will look into apache logs next.

By script timeout im guessing you mean max execution time?

closed 2024-03-18 by magicbug