magicbug/Cloudlog #2835: Top nav dropdowns are not expanding

patrickrb, 2023-12-14:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Dropdowns in the top nav bar do not expand/open in Edge

**To Reproduce**
Steps to reproduce the behaviour:
1. log in
2. try to hover over any dropdown in the top nav
3. unable to open a dropdown

**Expected behaviour**
Dropdown menus should expand

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser edge
 - Version  120.0.2210.61 (Official build) (arm64)

**Additional context**
I'm not seeing any warning or errors in the console.log

HB9HIL, 2023-12-14:
Press 

on Windows: "Ctrl + F5"

or

on Mac: "Cmd + Shift + R"

as it was described in the Version Info Popup.

Alternative would be to clear the Browser Cache.

We had a libary upgrade in this release which could cause this issue.

patrickrb, 2023-12-15:
that fixed it, thank you!

closed 2023-12-15 by patrickrb