magicbug/Cloudlog #2847: Blank page after update; Previously working UI

techfixpros, 2023-12-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Blank page when accessing Cloudlog/index.php. It has been working for several months, and CRON is set to update automatically. Guessing the latest version changed something. I am not using mod_rewrite and the .htacess.sample has not been renamed.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to http://bitnami-mariadb-1.lan/Cloudlog/ or http://bitnami-mariadb-1.lan/Cloudlog/index.php
2. Blank white page in browser
3. application/log shows `DEBUG - 2023-12-15 22:25:12 --> No URI present. Default controller set.
DEBUG - 2023-12-15 22:25:12 --> Migrating up from version 153 to version 154
ERROR - 2023-12-15 22:25:12 --> Severity: error --> Exception: Specified key was too long; max key length is 767 bytes /var/www/html/Cloudlog/system/database/drivers/mysqli/mysqli_driver.php 301`
4. I have cleared cache via ctrl+F5
**Expected behaviour**
UI shows up and is usable

**Screenshots**
Not applicable, its a blank page

**Desktop (please complete the following information):**
 - OS: Windows10
 - Browser Firefox
 - Version 120.0.1 (current latest)

**Smartphone (please complete the following information):**
 - Device: Galaxy Note
 - OS: [e.g. Android 13
 - Browser Firefox
 - Version 120.1.1

**Additional context**
Working for months without changing configuration. Nightly update may have caused issue. No changes have been made to mariaDB, but upon experiencing this issue I have restarted the DB.

magicbug, 2023-12-15:
What database version

techfixpros, 2023-12-15:
mariadb  Ver 15.1 Distrib 10.1.48-MariaDB, for debian-linux-gnu (x86_64) using readline 5.2

phl0, 2023-12-15:
Can you try replacing the 'constraint' => '255' in application/migration/154_create_dxpedition_table.php by 'constraint' => '32' for a test?

techfixpros, 2023-12-16:
I replaced it in all 3 instances and it looks to be working again.

techfixpros, 2023-12-16:
`DEBUG - 2023-12-16 01:07:26 --> Migrating up from version 153 to version 154                                                                                           x
DEBUG - 2023-12-16 01:07:26 --> Migrating up from version 154 to version 155                                                                                           x
DEBUG - 2023-12-16 01:07:26 --> Migrating up from version 155 to version 156                                                                                           x
DEBUG - 2023-12-16 01:07:26 --> Migrating up from version 156 to version 157                                                                                           x
DEBUG - 2023-12-16 01:07:27 --> Migrating up from version 157 to version 158                                                                                           x
DEBUG - 2023-12-16 01:07:27 --> Migrating up from version 158 to version 159                                                                                           x
DEBUG - 2023-12-16 01:07:27 --> Migrating up from version 159 to version 160                                                                                           x
DEBUG - 2023-12-16 01:07:27 --> Migrating up from version 160 to version 161                                                                                           x
DEBUG - 2023-12-16 01:07:27 --> Finished migrating to 161                                                                                                              x
DEBUG - 2023-12-16 01:07:27 --> Migration class already loaded. Second attempt ignored.                                                                                x
DEBUG - 2023-12-16 01:07:30 --> Migration class already loaded. Second attempt ignored.                                                                                x
DEBUG - 2023-12-16 01:08:59 --> You did not select a file to upload.                                                                                                   x
DEBUG - 2023-12-16 01:09:36 --> Electronicqsl class already loaded. Second attempt ignored.     `

closed 2023-12-16 by techfixpros