magicbug/Cloudlog #2877: LoTW import does not update QSO records if the time does not match precisely

jstrahle, 2023-12-25:
**Describe the bug**
I noticed that some of the QSO records are not being updated even when doing manual import from LoTW (tested with both pull data and adif-import options). While doing the update, it states that a matching record was found and it is being updated:
[image]

When I go and check theQSO log  record, it is not updated:
[image]

This is the QSO information from LoTW website:
[image]

I noticed that QSO time in my log and LoTW are having 1min difference. I tought that it might be causing the issue. I changed the entry in my log to be exactly the same as it is in LoTW. After doing this, the log entry updated when I re-run the LoTW import.

**Expected behaviour**
There should be some kind of tolerance with the QSO record time. LoTW import recognizes records to be updated, but the record is not being updated.

phl0, 2023-12-25:
Have the QSOs been uploaded from this Cloudlog instance to LoTW?

jstrahle, 2023-12-25:
@phl0 No, those log entries have been uploaded from different log software before I started using the Cloudlog

phl0, 2023-12-25:
> @phl0 No, those log entries have been uploaded from different log software before I started using the Cloudlog

Could you pls show an extract of the mentioned QSO in the other logging software that was used to upload to LoTW? What times does it show?

jstrahle, 2023-12-25:
> Could you pls show an extract of the mentioned QSO in the other logging software that was used to upload to LoTW? What times does it show?

Sure. There are actually two timestamps in Amateur Contact Log that I used before Cloudlog: Start and End. Looks like LoTW is using Start timestamp in their logs and after ADIF import, Cloudlog is only showing the End time.

[image]

jstrahle, 2023-12-25:
I made an ADIF export from Cloudlog and manually uploaded that via LoTW website, and after that did LoTW import in Cloudlog. Now it looks better. Will close the bug report because I believe that this behavior is related to chaning from log software to another.

closed 2023-12-25 by jstrahle

phl0, 2023-12-25:
Maybe. Would you be able and willing to provide the ADIF file sonthat I can have a look on what and how its imported?