magicbug/Cloudlog #2901: Get data from QRZ.com fails

hugosilvaparagramanet, 2024-01-02:
Get data from QRZ.com fails for some callsigns - verified In Advanced Logbook > Update from Callbook, Post QSO and Live QSO.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to either Live or Post QSO
2. Input callsign HB3YRZ/P
3. No data is returned

**Expected behaviour**
Data should be returned, user has QRZ.com information. Suspect that cause is bc user has callsign HB3YRZ that QRZ.com auto-redirects to HB3YRZ/P and the redirect is not being handled.

**Additional context**
Self-hosted Cloudlog

magicbug, 2024-01-03:
Your correct shows up OK based on his home data in the search box under the map but form fields aren't filled in, guess for /p in this case it should only be their name

hugosilvaparagramanet, 2024-01-03:
Just created a PR with a retry if it's a compound call sign. Not great, but it's a start.

closed 2024-01-04 by magicbug