magicbug/Cloudlog #2903: Edit Menu broken

do4bmw, 2024-01-02:
i did a update today 

and now the Edit Button in the logbook overview ist "broken"

see screenshot ... its CUT in half ... i testet Opera / Edge / Firefox and Chrome ... all the same

[image]

do4bmw, 2024-01-02:
[image]

magicbug, 2024-01-02:
I'd recommend doing a CTRL + F5 (clear browser cache) to see if it resolves it, else I'll dig into it a bit more but this should be resolved

do4bmw, 2024-01-02:
> I'd recommend doing a CTRL + F5 (clear browser cache) to see if it resolves it, else I'll dig into it a bit more but this should be resolved

that is what i tryed before i posted in here :-)

magicbug, 2024-01-02:
Can you confirm what theme you are using

do4bmw, 2024-01-02:
there is no change when i switch themes

[image]

at first i had Darkly , and this screenshot is the theme Cosmos

do4bmw, 2024-01-02:
same with "Cyborg"

[image]

magicbug, 2024-01-03:
Strange, I cant seem to replicate this either don't he default theme or cyborg

[image]

I'd check when you updated Cloudlog that ALL the files actually updated

DK9JC, 2024-01-03:
i have the same issue since 2.6.0 now also with 2.6.2.
cache of course cleared

[image]

DK9JC, 2024-01-03:
Just opened up another clean install of Firefox and it worked without any issue. 100% a local cache problem.

[image]

do4bmw, 2024-01-03:
[image]

i cleared all the cache but still have the problem :-(

do4bmw, 2024-01-03:
[image]

strange :-/

do4bmw, 2024-01-03:
i also tryed "Inkognito" Mode on the browser ...

but no change

do4bmw, 2024-01-03:
next i will cleanup "Windows" , maybe it helps

do4bmw, 2024-01-03:
ok windows "Dateisystembereinigung" inkl. clearing of all temp files = no change

i am running out of ideas

[image]

do4bmw, 2024-01-03:
Ok , one more Update , i just tryed it on my Raspberry PI = same problem

so it has nothing todo with the local cache

[image]

do4bmw, 2024-01-03:
> Just opened up another clean install of Firefox and it worked without any issue. 100% a local cache problem.
> 
> ![2024-01-03_17h09_47](https://private-user-images.githubusercontent.com/23286699/293967971-5e612058-05a4-4aee-ac1c-de5934518a02.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMDU1OTgsIm5iZiI6MTcwNDMwNTI5OCwicGF0aCI6Ii8yMzI4NjY5OS8yOTM5Njc5NzEtNWU2MTIwNTgtMDVhNC00YWVlLWFjMWMtZGU1OTM0NTE4YTAyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAzVDE4MDgxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlMjNjNGQ1Zjc2OWJmMzQ5NmFjNzc3MjBiOGM1ZmMxZTljMmNjYTcwOWJkNmFlOTY4YjE2ZjQxNGIyODA5OTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Vv_YSqQ5b9oLkBQW9ydSakeqGjni-hR3ar2ECz4IW1k)

i see a difference ... your menu opens to the left
my is trying to the right site ...

do4bmw, 2024-01-04:
if the menu opens to left side , then there is no problem ...
but on my installation , since the update a few days ago , it will open to the right side
maybe there is point in the code to change that , if yes , were it is ... i am not a coding expert

do4bmw, 2024-01-04:
topic can be closed

i made the Update today to 2.6.2 , and the error is gone

many thanks