magicbug/Cloudlog #2915: Error Message in "view QSL Cards"

vbueschken, 2024-01-09:
Today I have installed Cloudlog on my hosted environment
PHP 8.2 

When I click on "view QSL Cards" I get following error message:

An uncaught Exception was encountered
Type: InvalidArgumentException
Message: where_in() expects $values to be a non-empty array
Filename: /www/htdocs/buesccom/cloudlog.bueschken.com/system/database/DB_query_builder.php
Line Number: 884

magicbug, 2024-01-10:
Was this a fresh install with no Qsos present?

Message: where_in() expects $values to be a non-empty array

Suggests that no Logbook has been created with an attached Station Location

vbueschken, 2024-01-10:
Yes..  this is a fresh install... no data.Gesendet von Outlook für Android

magicbug, 2024-01-10:
Perfect, thanks that needs cleaned up :)

magicbug, 2024-01-10:
Not a bug per say but visually not very nice!

vbueschken, 2024-01-10:
I also get this Message, when I try to save a Live QSO:

A PHP Error was encountered
Severity: 8192
Message: trim(): Passing null to parameter #1 ($string) of type string is deprecated
Filename: models/Logbook_model.php
Line Number: 297

May be the same cause..?
The Record is saved - I can see it in Logbook Advanced

magicbug, 2024-01-10:
> I also get this Message, when I try to save a Live QSO:
>
> A PHP Error was encountered Severity: 8192 Message: trim(): Passing null to parameter #1 ($string) of type string is deprecated Filename: [...]

This is unrelated i'd suggest creating a new  issue but perhaps you haven't created a station location and linked it to a Station Logbook?

commit 6dde203b (2024-03-18): Fixes #2915
closed 2024-03-18 by magicbug