magicbug/Cloudlog #2916: Option wheel dropdown items are cutoff

Reid-n0rc, 2024-01-09:
**Describe the bug**
From the Log Overview page, when I click the option wheel the items under it are cutoff as shown in the screenshot below.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Logbook -> Overview'
2. Click on 'Option wheel' to the right of any contact
4. See error

**Expected behaviour**
The Option wheel items should be visible

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10
 - Browser Chrome Version 120.6099.200
 - Cloudlog Version 2.6.2

magicbug, 2024-01-09:
Please make sure you have cleared your browser cache recently and try again

Reid-n0rc, 2024-01-09:
> Please make sure you have cleared your browser cache recently and try again

I cleared the cache and restarted Chrome. Same result.

Reid-n0rc, 2024-01-09:
I should add that the menu will move/resize when the option wheel is clicked. The screenshot shows the result of the mouseover.

magicbug, 2024-01-09:
> I cleared the cache and restarted Chrome. Same result.

How did you clear the cache? CTRL+F5? if that didn't fix it please carry out clearing cache using https://support.google.com/accounts/answer/32050?hl=en&co=GENIE.Platform%3DDesktop

Reid-n0rc, 2024-01-09:
> > I cleared the cache and restarted Chrome. Same result.
> 
> How did you clear the cache? CTRL+F5? if that didn't fix it please carry out clearing cache using https://support.google.com/accounts/answer/32050?hl=en&co=GENIE.Platform%3DDesktop

This is the method I used. Performed it again per linked instructions. Did `CTRL+F5`. This give the same result on mouseover. Again, actually clicking the option wheel causes the dropdown text to move and become readable for that log entry until the page is refreshed.

Reid-n0rc, 2024-01-09:
@magicbug just found that clicking the option wheel again causes the text to move back to the unreadable position.

shortcircuitbtc, 2024-01-10:
I can confirm this behavior on my instance.

lowUser, 2024-01-10:
I can confirm this behavior.

cats-shadow, 2024-02-05:
> How did you clear the cache? 

I had the same problem. Clearing the browser cache (Firefox Win64) helped.

Now there is a funny effect (can be achieved on purpose, I don't know if it is considered a bug).
If you click on the gear, then move the cursor away and bring it back to the same gear you clicked on, the popup menu is not displayed. See the screenshot below. The row is highlighted, but there is no menu, the cursor is over the gear.

[image]