magicbug/Cloudlog #2928: PHP Error when trying to import LOTW certificate

raufaser, 2024-01-20:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
2 PHP errors are thrown when trying to import a LOTW certifiacte

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Log Of The World'
2. Click on 'Upload Certificate'
3. Click on 'Choose File'
4. Pick the .p12 file from the file picker dialog
5. Click on 'Upload File'
6. See error

```
A PHP Error was encountered
Severity: Warning
Message: Undefined array key "cert"
Filename: controllers/Lotw.php
Line Number: 407

A PHP Error was encountered
Severity: Warning
Message: Cannot modify header information - headers already sent by (output started at /var/www-cloudlog/system/core/Exceptions.php:283)
Filename: helpers/url_helper.php
Line Number: 564
```

**Expected behaviour**
The .p12 LOTW certificate is uploaded to cloudlog

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Thorium
 - Version Version 111.0.5563.111

**Additional context**
The certificate has NO password.
The installation is a normal installation. NO docker.

raufaser, 2024-01-27:
Thanks for pointing me to the log.

The interesting lines: (more in attachement)
[log.txt](https://github.com/magicbug/Cloudlog/files/14072152/log.txt)

```
ERROR - 2024-01-27 11:44:52 --> Severity: Warning --> Undefined array key "cert" /var/www-cloudlog/application/controllers/Lotw.php 407
ERROR - 2024-01-27 11:44:52 --> error:0308010C:digital envelope routines::unsupported
ERROR - 2024-01-27 11:44:52 --> Severity: Warning --> Cannot modify header information - headers already sent by (output started at /var/www-cloudlog/system/core/Exceptions.php:283) /var/www-cloudlog/system/helpers/url_helper.php 564
```

I will investigate this further `error:0308010C:digital envelope routines::unsupported`

Seems to be something open_ssl / php related.

Cloudlog is running on a VPS:

- Ubuntu 22.04.3 LTS
- Kernel 5.15.0-91-generic
- nginx/1.25.3
- PHP 8.2.13
- mariadb  Ver 15.1 Distrib 10.11.6-MariaDB

magicbug, 2024-01-27:
That helps are you exporting the certficiate from tQSL using the latest version?

raufaser, 2024-01-27:
> That helps are you exporting the certficiate from tQSL using the latest version?

No I wasn't. And that was the problem. I wouldn't not have guessed that this could be an issue. My version wasn't ancient. 2.6.something ...

Anyways. Thanks a lot for the help and your awesome work in general.

closed 2024-01-27 by raufaser