magicbug/Cloudlog #2929: It seems that the eQSL site has changed. Please open up an issue on GitHub.

HochdruckHummer, 2024-01-21:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**

It seems that the eQSL site has changed. Please open up an issue on GitHub. -> Shown when trying to upload to eqsl.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to user, hover above it
2. Click on 'eQSL Import/Export'
3. Choose 'Import directly from eQSL'
4. Click 'Import eQSL QSO matches
5. see error

**Expected behaviour**
Upload

phl0, 2024-01-21:
Cannot confirm here. Still working as it should. Please provide more details (e.g. excerpt from log file).

magicbug, 2024-01-21:
Yeah cant replicate here just tried it on two copies so errors would be useful

HochdruckHummer, 2024-01-21:
Where do I find the logs myou need in directory structure?

magicbug, 2024-01-21:
PHP error logs /var/log relating to the server if its linux.. else enable logging by going into /application/config/config.php and then you'll get logs in /application/logs when you use Cloudlog which will tell us what went wrong

phl0, 2024-01-21:
Related to https://github.com/magicbug/Cloudlog/issues/2926 maybe?

magicbug, 2024-01-21:
Ah yes definitely related to that

HochdruckHummer, 2024-01-21:
It can't be in connection with my former issue, which I closed. I was not able to get it running on my webhoster, so I installed WAMP on WinServer and everything works now. 

Anyways, I still call cloudlog by external domain and subdomain...

closed 2024-02-05 by magicbug