magicbug/Cloudlog #2940: Reverse Proxy causes HTTP 307

stevendodd, 2024-01-28:
Everything works great except for when I put it behind a reverse proxy. Here is my apache reverse proxy config:

  ProxySourceAddress <proxy ip>
  ProxyPass /cloudlog http://<cloudlog server ip>
  ProxyPassReverse /cloudlog http://<cloudlog server ip>
  ProxyPassReverseCookieDomain  "<cloudlog server ip>"  "<external domain>"
  ProxyPassReverseCookiePath  "/"  "/cloudlog/"

config.php:
$config['proxy_ips'] = '<proxy ip>';

I've tried all sorts of combinations and just can't get it to work; it keeps on doing an http 307 redirect when going via the proxy and the pages don't load properly externally.

magicbug, 2024-01-28:
I dont use it behind a proxy so sadly cant offer any help, you might get some help if you stick it in discussions as this isn't a code issue as such

stevendodd, 2024-01-30:
https://github.com/magicbug/Cloudlog/discussions/2941#discussioncomment-8310270

closed 2024-01-30 by stevendodd

stevendodd, 2024-01-30:
Adding this in config.php means that both local and proxy work

```
if ($_SERVER['REMOTE_ADDR'] == '<proxy ip address>') {
  $config['base_url'] = 'https://<external domain>/cloudlog/';
} else {
  $config['base_url'] = 'http://<cloudlog server ip address>';
}
```